grigorig/stcgal #1: STC15L2K60S2

blindman2k, 2015-02-02:
Hi Grigori,

Your STC programmer is the first one I have found that comes really close to programming my STC15L2K60S2. It doesn't succeed but at least it correctly detects it. Are you able to assist to get it live? I am getting slightly desperate.

Yours,
    A.

datacompboy, 2015-02-02:
Can you show used command line optinos and full running output?

blindman2k, 2015-02-02:
Hi and thanks datacompboy,

```
$ ./stcgal.py -P stc15 -p /dev/cu.usbserial-A703BXNC doc/hello.bin
Waiting for MCU, please cycle power: done
Target model:
  Name: STC15L2K60S2
  Magic: F488
  Code flash: 60.0 KB
  EEPROM flash: 1.0 KB
Target frequency: 169.245 MHz
Target BSL version: 7.2S
Target options:
  reset_pin_enabled=False
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=8
  low_voltage_reset=True
  low_voltage_threshold=0
  eeprom_lvd_inhibit=False
  eeprom_erase_enabled=True
  bsl_pindetect_enabled=True
Trimming frequency: Protocol error: incorrect frame start
Disconnected!
```

grigorig, 2015-02-03:
Unfortunately, the STC15L2/F2 etc. parts use yet another protocol variant different from the STC15F104E protocol that I reverse engineered. I didn't yet have time to investigate what's going on. It looks like it uses a quite different method to tune the internal RC oscillator for proper UART clock. Like the other protocol variants it's probably just as strange and convoluted.

Yeah, STC sucks. If you want to give reversing this a try, use a serial port sniffer (I simply used two cheap UART-USB thingies to monitor RX and TX) and jpnevulator.

blindman2k, 2015-02-03:
I already have a number of captures but haven't got the skills to reverse
engineer it.
STC is a right royal pain :(
Would you be interested in a few free chips and a bit of paid consulting
work to do it?

datacompboy, 2015-02-04:
Can you send me your several traces you have?

blindman2k, 2015-02-04:
Sure.
https://www.dropbox.com/s/wbsdothzbq3l05k/stc_logs.zip

blindman2k, 2015-02-13:
@datacompboy, @grigorig 

Have either of you been able to crack the code? Would you be willing to give it a shot?
I am happy to send you some chips (chicken flavour rocks).

datacompboy, 2015-02-13:
Sorry, i have scheduled that on this sunday-monday, and will write some info there asap.
If you have more logs with another binary uploaded -- it can be helpful.

blindman2k, 2015-02-13:
I can't get any more logs till I get in front of the hardware on Monday. Sorry.

Thanks heaps for helping out. We are starting to look at redesigning the PCB in order to replace the chip with something programmable. It will cost $$$s if we don't find a solution.

blindman2k, 2015-02-16:
Here are some logs with a new version of the firmware.
https://www.dropbox.com/s/z0xv48ya0v0mf0r/stc_logs2.zip

I am told that the code below runs on an STC chip and will program another STC chip over UART.
I can't read the code or comments enough to understand if that is true or not but it might prove useful.
https://www.dropbox.com/s/5nzc7qdj8u5iha6/stcisp_c.zip

blindman2k, 2015-02-16:
That code seems to work quite well. I am just tweaking the user code writing to get it to work but it looks like this is the solution.

grigorig, 2015-02-17:
The code looks interesting. Where is that from? Maybe it's time to finally
play around with this a little more.

blindman2k, 2015-02-17:
It came from our CM but its findable on STC's website via Google: http://www.stcisp.com/mcuisp_stc15f_code_c.html
The translation was from Google Translate.
The version on the link above is actually a newer version than I attached and worked better.

I managed to get everything working except the Hardware Options section at the end which sets the frequency. Thankfully, I don't need to use it. So I am all fully operational just by porting this code.

grigorig, 2015-11-20:
That's very helpful, although that code completely skips over the crucial parts, i.e. frequency calibration and option bits. Anyway, I have mostly reverse engineered the frequency calibration and adjustment now. If you still care, I'll probably push some support for the chips with BSL version 7.x soon. Maybe it's time for a release, I think all current chips should be supported now.

grigorig, 2015-11-22:
Support for these devices has been added and should be working fine.

closed 2015-11-22 by grigorig