grigorig/stcgal #3: Success report with STC15L2K16S2

papillon81, 2015-11-23:
Hi,

just want to report a successful read with the latest version as of now and a STC15L2K16S2:

```
./stcgal.py -P stc15  -p /dev/ttyAMA0 -D
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 34 50 6C EA 7C 91 F7 BA 9E 00 A8 D7 70 FD 27 02 FF FF 72 53 00 F4 82 04 06 70 91 02 18 1C 21 23 24 00 F4 40 04 DB FF 7C BF FF FF 15 07 21 51 15 6E 16
done
Target model:
  Name: STC15L2K16S2
  Magic: F482
  Code flash: 16.0 KB
  EEPROM flash: 45.0 KB
Target frequency: 23.966 MHz
Target BSL version: 7.2S
Target wakeup frequency: 27.882 KHz
Target options:
  reset_pin_enabled=False
  clock_source=external
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=2
  eeprom_lvd_inhibit=True
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart2_passthrough=False
  uart2_pin_mode=normal
-> Packet data: 46 B9 6A 00 07 82 00 F3 16
Disconnected!
```

Is there a way to actually read the flash content from a chip?

grigorig, 2015-11-23:
Thank you for the report! So BSL 7.2 works too. I don't have a part with that version. I'll add it to the compatibility list.

Unless there are bugs in the BSL, I don't think it is possible to read back flash content. It's supposed to be a security feature. STC or their customers seem to be extremely paranoid when it comes to people "stealing" code. Newer chips have some strange encryption features in the BSL as well. (I don't think stcgal will ever support these, but who cares?)

grigorig, 2015-11-23:
Oh, btw, I see you just did a device query, can you check if programming actually works?

papillon81, 2015-11-24:
I would love to try it out. However, as long as I can't backup the original flash contents, I would brick the device. Do you have any directions for me where to look for a way to read the contents? Maybe there is no protection in this device.

grigorig, 2015-11-24:
Like I said, as far as I'm aware, the BSL doesn't support it. It is write-only. Your best bet is a bug in the firmware or BSL that allows you to read arbitrary bytes from flash memory.

grigorig, 2015-11-24:
Note, if you don't have a backup of the original firmware, of course I do not expect that you try this. I have some more STC MCUs being sent to me right now for testing purposes anyway.

papillon81, 2015-11-26:
I re-considered this. As the device with the STC chip is broken anyway and I got it for free now, i'll try to flash it. As i do not have a IDE installed, can you supply me with a hex file that I can use for testing?

papillon81, 2015-11-27:
I did it myself, set up an IDE and just flashed a "hello world" program:

```
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 34 50 6C EA 7C 91 F7 BA 9E 00 A8 D7 70 FD 4E 0D FF FF 72 53 00 F4 82 04 06 70 91 02 18 1C 21 23 24 00 F4 40 04 DB FF 7C BF FF FF 15 07 21 51 15 A0 16
done
Target model:
  Name: STC15L2K16S2
  Magic: F482
  Code flash: 16.0 KB
  EEPROM flash: 45.0 KB
Target frequency: 23.977 MHz
Target BSL version: 7.2.4S
Target wakeup frequency: 27.882 KHz
Target options:
  reset_pin_enabled=False
  clock_source=external
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=2
  eeprom_lvd_inhibit=True
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart2_passthrough=False
  uart2_pin_mode=normal
Loading flash: 105 bytes (Intel HEX)
Switching to 19200 baud: -> Packet data: 46 B9 6A 00 0E 01 91 40 FE C6 00 00 81 03 8F 16
<- Packet data: 46 B9 68 00 07 01 00 70 16
-> Packet data: 46 B9 6A 00 0B 05 00 00 5A A5 01 79 16
<- Packet data: 46 B9 68 00 07 05 00 74 16
done
Erasing flash: -> Packet data: 46 B9 6A 00 0B 03 00 00 5A A5 01 77 16
<- Packet data: 46 B9 68 00 0E 03 F4 82 00 99 01 A4 BF 03 EC 16
done
Writing 256 bytes: -> Packet data: 46 B9 6A 00 4B 22 00 00 5A A5 02 00 06 02 00 62 75 81 07 12 00 65 E5 82 60 03 02 00 03 79 00 E9 44 00 60 1B 7A 00 90 00 69 78 01 75 A0 00 E4 93 F2 A3 08 B8 00 02 05 A0 D9 F4 DA F2 75 A0 FF E4 78 7F F6 D8 FD 78 00 E8 44 00 1B 11 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
.-> Packet data: 46 B9 6A 00 4B 02 00 40 5A A5 60 0A 79 01 75 A0 00 E4 F3 09 D8 FC 78 00 E8 44 00 60 0C 79 00 90 00 01 E4 F0 A3 D8 FC D9 FA 02 00 03 80 FE 22 75 82 00 22 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 13 99 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
.-> Packet data: 46 B9 6A 00 4B 02 00 80 5A A5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 36 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
.-> Packet data: 46 B9 6A 00 4B 02 00 C0 5A A5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 76 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
. done
Finishing write: -> Packet data: 46 B9 6A 00 0B 07 00 00 5A A5 01 7B 16
<- Packet data: 46 B9 68 00 08 07 54 00 CB 16
 done
Setting options: -> Packet data: 46 B9 6A 00 4B 04 00 00 5A A5 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 01 FF 6E FF 36 FF 00 FF FD FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 91 7F F7 BA 9E 3C 83 16
<- Packet data: 46 B9 68 00 08 04 54 00 C8 16
done
Target UID: F482009901A4BF
-> Packet data: 46 B9 6A 00 07 82 00 F3 16
Disconnected!
```

grigorig, 2015-11-27:
Looks good. Sorry, I did not have any time yet to prepare a nice testing program (e.g. something that toggles outputs based on some delay, or something like that). Does your "hello world" program work correctly?

zerog2k, 2015-11-27:
I started a repo for simple blinky example w/ stc15F204EA (should work with 15F/Lxxx, im guessing) if this is useful for testing:
https://github.com/zerog2k/stc_blinky

papillon81, 2015-11-29:
I took the blinky code and adapted it a bit. I can now confirm that the flashing works fine here. Nice work!

closed 2015-11-29 by papillon81