grigorig/stcgal #6: stc15f104w issue

Phil67ago, 2016-01-03:
Hi!
Seems that stc15f104w is different than a standard stc15.

Did some effort to figure it out... but I failed!
status_packet is one byte too short

--

#### stcgal -p /dev/ttyUSB0 -D -P stc15 binary.hex

Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 2B 50 91 C5 92 B5 F7 BB 9F 00 A8 E0 D0 FF 4E 78 00 00 71 51 03 F2 94 04 06 58 B5 02 1B 20 22 26 22 80 14 10 04 CF 0F 6E 16
('/usr/bin/stcgal', 9, '<module>', ["        load_entry_point('stcgal==1.0', 'console_scripts', 'stcgal')()\n"], 0)
('/usr/lib/python3.4/site-packages/stcgal-1.0-py3.4.egg/stcgal/frontend.py', 184, 'cli', ['    return gal.run()\n'], 0)
('/usr/lib/python3.4/site-packages/stcgal-1.0-py3.4.egg/stcgal/frontend.py', 120, 'run', ['        try: self.protocol.connect()\n'], 0)
('/usr/lib/python3.4/site-packages/stcgal-1.0-py3.4.egg/stcgal/protocols.py', 761, 'connect', ['                status_packet = self.get_status_packet()\n'], 0)
('/usr/lib/python3.4/site-packages/stcgal-1.0-py3.4.egg/stcgal/protocols.py', 1454, 'get_status_packet', ['        status_packet = self.read_packet()\n'], 0)
('/usr/lib/python3.4/site-packages/stcgal-1.0-py3.4.egg/stcgal/protocols.py', 1231, 'read_packet', ['        self.dump_packet(packet, receive=True)\n'], 0)
('/usr/lib/python3.4/site-packages/stcgal-1.0-py3.4.egg/stcgal/protocols.py', 655, 'dump_packet', ['            for item in reversed(inspect.stack()):\n'], 0)
done
Target model:
  Name: STC15F104W
  Magic: F294
  Code flash: 4.0 KB
  EEPROM flash: 1.0 KB
Target frequency: 11.068 MHz
Target BSL version: 7.1.4Q
Target wakeup frequency: 37.317 KHz
len=37, 50 91 C5 92 B5 F7 BB 9F 00 A8 E0 D0 FF 4E 78 00 00 71 51 03 F2 94 04 06 58 B5 02 1B 20 22 26 22 80 14 10 04 CF
F7 BB 9F FF
Traceback (most recent call last):
  File "/usr/bin/stcgal", line 9, in <module>
    load_entry_point('stcgal==1.0', 'console_scripts', 'stcgal')()
  File "/usr/lib/python3.4/site-packages/stcgal-1.0-py3.4.egg/stcgal/frontend.py", line 184, in cli
    return gal.run()
  File "/usr/lib/python3.4/site-packages/stcgal-1.0-py3.4.egg/stcgal/frontend.py", line 120, in run
    try: self.protocol.connect()
  File "/usr/lib/python3.4/site-packages/stcgal-1.0-py3.4.egg/stcgal/protocols.py", line 771, in connect
    self.initialize_options(status_packet)
  File "/usr/lib/python3.4/site-packages/stcgal-1.0-py3.4.egg/stcgal/protocols.py", line 1662, in initialize_options
    self.options = Stc15Option( status_packet[5:8] + status_packet[12:13] + status_packet[37:38])
  File "/usr/lib/python3.4/site-packages/stcgal-1.0-py3.4.egg/stcgal/protocols.py", line 452, in __init__
    assert len(msr) == 5
AssertionError

---------------------------------------------------

#### stcgal -p /dev/ttyUSB0 -D binary.hex

Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 2B 50 91 C5 92 B5 F7 BB 9F 00 A8 E0 D0 FF 4E 78 00 00 71 51 03 F2 94 04 06 58 B5 02 1B 20 22 26 22 80 14 10 04 CF 0F 6E 16
done
Target model:
  Name: STC15F104W
  Magic: F294
  Code flash: 4.0 KB
  EEPROM flash: 1.0 KB
Target frequency: 76.068 MHz
Target BSL version: 7.1Q
Target options:
  reset_pin_enabled=False
  low_voltage_reset=True
  oscillator_stable_delay=4096
  por_reset_delay=long
  clock_gain=high
  clock_source=internal
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=64
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=True
Loading flash: 249 bytes (Intel HEX)
Switching to 19200 baud: -> Packet data: 46 B9 6A 00 0D 50 00 00 36 01 F2 94 02 84 16
Serial port error: read timeout

---

/Phil

zerog2k, 2016-01-03:
Try -P stc15a
This changed for me recently as well on stc15f204ea

Phil67ago, 2016-01-04:
Same serial timeout as without....

#### stcgal -p /dev/ttyUSB0 -D -P stc15a binary.hex
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 2B 50 91 C5 92 B5 F7 BB 9F 00 A8 C9 60 FF 4E 78 69 6E 71 51 03 F2 94 04 06 58 B5 02 1B 20 22 26 22 80 14 10 04 CF 0F BE 16
done
Target model:
  Name: STC15F104W
  Magic: F294
  Code flash: 4.0 KB
  EEPROM flash: 1.0 KB
Target frequency: 92.056 MHz
Target BSL version: 7.1Q
Target options:
  reset_pin_enabled=False
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=64
  low_voltage_reset=True
  low_voltage_threshold=0
  eeprom_lvd_inhibit=False
  eeprom_erase_enabled=True
  bsl_pindetect_enabled=True
Loading flash: 583 bytes (Intel HEX)
Trimming frequency: -> Packet data: 46 B9 6A 00 0D 50 00 00 36 01 F2 94 02 84 16
Serial port error: read timeout

zerog2k, 2016-01-04:
Hmm. Looks like indeed -P stc15 gave you the most reasonable result (detected clock freq properly).
What kind of uart-usb adapter are you using?
Perhaps @grigorig will need to assist.

grigorig, 2016-01-04:
stc15 should be the correct protocol for STC15F104W. This is probably a regression. I have an STC15L104W with the same BSL version, let me check.

closed 2016-01-05 by grigorig