grigorig/stcgal #9: Unable to upload to STC12C5204AD

rfb, 2016-03-18:
Linux, CH340 UART, pySerial 3.0.1. I don't have any alternative UARTs at the moment unless I can figure out how to make this work with a bus pirate.

I've tried different baud rates from 9600 to 57600, matching handshake and protocol baud, and the alternative stc12a protocol. Haven't found any combination that works so far.

Is it possible to write protect STC MCUs?

Waiting for MCU, please cycle power: done
Target model:
  Name: STC12C5204AD
  Magic: E164
  Code flash: 4.0 KB
  EEPROM flash: 2.0 KB
Target frequency: 11.981 MHz
Target BSL version: 6.6H
Target options:
  reset_pin_enabled=True
  low_voltage_reset=True
  oscillator_stable_delay=32768
  por_reset_delay=short
  clock_gain=high
  clock_source=external
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=64
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
Loading flash: 104 bytes (Intel HEX)
Switching to 19200 baud: Serial port error: read timeout

grigorig, 2016-03-18:
> Is it possible to write protect STC MCUs?

Sort of. You can set a "password", but this only supported on STC 15 parts with somewhat new BSL. So not a problem in this case.

Your part uses the STC12 protocol which uses parity (8E1) for safer UART communication. The Linux ch340/ch341 driver does not support parity at all (!). So it cannot work.

I've done some work to fix the deficiency of the driver, please try again with the patches in  https://github.com/grigorig/linux/commits/ch340

rfb, 2016-03-19:
> Your part uses the STC12 protocol which uses parity (8E1) for safer UART communication. 

Is that in the STC12C5204AD datasheet?

I'll give the linux patches a shot.

grigorig, 2016-03-19:
No, the datasheets don't document the bootloader protocol at all. I reverse engineered it.

rfb, 2016-03-20:
That worked beautifully, thank you! 

FYI: I built the patch against 4.3.5, which is the current kernel shipped with debian/stretch. No error or warnings for the module during compile. Nothing of interest in dmesg when the ch340 is plugged in.

```
Waiting for MCU, please cycle power: done
Target model:
  Name: STC12C5204AD
  Magic: E164
  Code flash: 4.0 KB
  EEPROM flash: 2.0 KB
Target frequency: 11.981 MHz
Target BSL version: 6.6H
Target options:
  reset_pin_enabled=True
  low_voltage_reset=True
  oscillator_stable_delay=32768
  por_reset_delay=short
  clock_gain=high
  clock_source=external
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=64
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
Loading flash: 104 bytes (Intel HEX)
Switching to 19200 baud: testing setting done
Erasing 2 blocks: done
Writing 256 bytes: .. done
Finishing write: done
Setting options: done
Target UID: E164011501D291
Disconnected!
```

closed 2016-03-20 by rfb

grigorig, 2016-03-20:
Nice, so it seems to work. :)

Just a short request, can you please tell me what chip version you have? You have to enable debug output for that. If you have dynamic debug enabled in your kernel configuration (which is likely the case), reloading ch341 with debug enabled should do it:

    modprobe -r ch341
    modprobe ch341 dyndbg==pmf

Also, can I include your name in a Tested-By line in the git commit?

rfb, 2016-03-20:
It looks like chip version 48.

> Also, can I include your name in a Tested-By line in the git commit?

Yes of course, 

```
[ 6295.705483] usbcore: registered new interface driver usbserial
[ 6295.705505] usbcore: registered new interface driver usbserial_generic
[ 6295.705526] usbserial: USB Serial support registered for generic
[ 6295.705892] usbcore: registered new interface driver ch341
[ 6295.705906] usbserial: USB Serial support registered for ch341-uart
[ 6304.357980] usb 1-1: new full-speed USB device number 7 using xhci_hcd
[ 6304.487439] usb 1-1: New USB device found, idVendor=1a86, idProduct=7523
[ 6304.487442] usb 1-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 6304.487444] usb 1-1: Product: USB2.0-Serial
[ 6304.487862] ch341 1-1:1.0: ch341-uart converter detected
[ 6304.487906] ch341:ch341_control_in: usb 1-1: ch341_control_in(c0,5f,0000,0000,ffff880213da8620,2)
[ 6304.487963] ch341:ch341_configure: usb 1-1: ch341_configure - chip version: 48
[ 6304.487967] ch341:ch341_control_out: usb 1-1: ch341_control_out(40,a1,0000,0000)
[ 6304.488011] ch341:ch341_control_in: usb 1-1: ch341_control_in(c0,95,2518,0000,ffff880213da8620,2)
[ 6304.488067] ch341:ch341_control_out: usb 1-1: ch341_control_out(40,9a,2518,00c3)
[ 6304.488109] ch341:ch341_control_in: usb 1-1: ch341_control_in(c0,95,0706,0000,ffff880213da8600,2)
[ 6304.488163] ch341:ch341_control_out: usb 1-1: ch341_control_out(40,a1,009c,b282)
[ 6304.488203] ch341:ch341_control_out: usb 1-1: ch341_control_out(40,a4,ff9f,0000)
[ 6304.488240] ch341:ch341_control_in: usb 1-1: ch341_control_in(c0,95,0706,0000,ffff880213da8600,2)
```

rfb, 2016-03-20:
Side note: I uploaded a simple firmware and I appear to have bricked the device.

```
void main(void) { while(1){} }
```

Any thoughts on what I might have done wrong?
  - Did I accidentally override the BSL? 
  - Are the Rx/Tx pins now set in a mode that prevents the BSL from work?

grigorig, 2016-03-20:
Are you sure it's bricked? To my knowledge, you cannot overwrite the BSL. Also note that you can also accidentally power the chip through any input pin (the clamping diodes, in fact). The bootloader is only invoked on power-on-reset so you need to be careful with that.

Something could be messed up with target option programming, though. There's an option that can be set (`bsl_pindetect_enabled`) which configures the bootloader so that you have to tie two pins to ground for the bootloader to be invoked, refer to the datasheet.

rfb, 2016-03-21:
Cool, tied P1.1/P1.0 to ground and it worked.

From the datahsheet:

> STC12C5205/5206, STC12LE5205/5206 it is required that P1.0/P1.1 is shorted to ground when users download their programs

grigorig, 2016-03-21:
Interesting. So that option is basically always enabled on certain MCUs. STC's hardware is so damn quirky...

rfb, 2016-03-22:
I figure someone in the future will find guidance from these tickets. 

No problem on the testing, I'm happy to do further testing if you would like. Just send me a message.