grigorig/stcgal #10: Support for USB-ISP

tpimh, 2016-03-22:
Will stcgal support USB-ISP?

[image]

grigorig, 2016-03-22:
I have an STC15W4 MCU, but I haven't tried the USB mode yet, sorry.

grigorig, 2016-05-12:
A correction: the USB-ISP mode doesn't use CDC-ACM, but raw control packets. The protocol seems to be quite simple, though. You can expect USB-ISP support, based on libusb/pyusb, to come.

grigorig, 2016-05-12:
WIP branch: https://github.com/grigorig/stcgal/tree/usbisp

grigorig, 2016-05-13:
The code in the branch is somewhat functional now. Flash programming works. If you are interested, please give it a go. The implementation requires pyusb.

tpimh, 2016-05-14:
Thank you very much! I'm very interested, but unfortunately I don't have any IC supporting USB-ISP right now to test it.

grigorig, 2016-05-14:
Setting options also works now. I'm confident this won't brick any chips, so it has been cleaned up and merged now.

closed 2016-05-14 by grigorig

wc7086, 2024-01-06:
@grigorig Can you add this support for stc8?

[image]