grigorig/stcgal #12: Unable to upload to STC12C5A60S2

rgm3, 2016-04-14:
GH issues aren't great for support, but I couldn't find a website or forum, apologies.  Thanks for making stcgal.  I recently received an STC12C5A60S2 as part of an LED cube kit and I'm [trying to program it myself](https://github.com/rgm3/ledcube444) on Linux Mint 17.3 through an FTDI adapter (Adafruit FTDI friend) set to 5V with DTR support.

```
stcgal -P stc12 -p /dev/ttyUSB0 -a
```

I can't seem to get my chip into bootloader mode, so stcgal says "Waiting..." forever.  Can you please help me understand how one might use external circuitry to leverage the DTR assertion support?  Currently I've hooked DTR up to pin 9 (RESET). I also tried the POR circuit shown on page 54 of the datasheet -- putting RESET between a 10 uF cap through a 10 Kohm resistor.

I tried to cut/restore power by unplugging and re-plugging my FTDI cable from the breadboard thinking that this is what was meant by power-on reset. No joy even if RESET pin high or low when power is applied.  If I only pull the 5V line the stc12 seems to be weakly powered through the RX pin (which is connected to the TX pin on my FTDI and transmitting via stcgal).

My python is 3.4.3 and python3-serial package is version 2.6-1build1.  I've double-checked my TX and RX lines, and also swapped them 'round just in case.  I tried "stc12a" protocol too.

grigorig, 2016-04-14:
The bootloader is only invoked by power-on-reset of the microcontroller. You have to toggle power to actually switch it off and on again. Then the bootloader will stand by and wait for the start sequence, but only for hundreds of milliseconds. The RESET pin does not do anything to start the bootloader. So please just keep RESET disconnected or use the datasheet recommended circuitry.

The DTR line is supposed to be used with some power switch (e.g. MOSFETs) to switch power to the microcontroller. Maybe I should clarify this. The parasitic power supply through the TX pin can indeed be a problem. To get around this, you can insert a series resistor (e.g. 10K) to the TX line.

So, switch off and on the microcontroller by disconnecting and connecting the power pin while stcgal waits for the bootloader to respond. If the parasitic powering is actually a problem, add a resistor.

rgm3, 2016-04-14:
Thank you, that leads me to believe that my intuition of plugging the FTDI cable into the breadboard should have worked, since i'm currently powering the uC through the FTDI's VCC line.  "Power-on reset" is a new phrase to me, and seemed ambiguous despite being regarded as obvious.  The datasheet does not help clarify it either.  To your credit, stcgal's prompt is clear - "please cycle power."

Can you confirm that STC12C5A60S2 uses the "stc12" protocol?

Unfortunately, despite trying a 10K on the TX line, I'm still unable to connect to the chip.  The only next step I can think of is to track down a Windows computer to verify functionality with the "official" software.

grigorig, 2016-04-14:
Yes, stc12 is the correct protocol.

rgm3, 2016-04-14:
Success!
```
stcgal -P stc12 -l 9600

Waiting for MCU, please cycle power: done
Target model:
  Name: STC12C5A60S2
  Magic: D17E
  Code flash: 60.0 KB
  EEPROM flash: 2.0 KB
Target frequency: 11.059 MHz
Target BSL version: 7.1I
Target options:
  reset_pin_enabled=True
  low_voltage_reset=False
  oscillator_stable_delay=32768
  por_reset_delay=long
  clock_gain=high
  clock_source=external
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
Disconnected!
```
Your responses were helpful and appreciated. Cheers!

closed 2016-04-14 by rgm3

grigorig, 2016-04-14:
But, what was the underlying problem? :) I'm curious.

rgm3, 2016-04-14:
Hacking at 2AM mostly was the problem, combined with inexperience and "Arduino-like" expectations.  Things went MUCH faster after you helped me realize POR is simply "unplug it and plug it back in" and not an arcane pin toggling / external circuit procedure. I stopped trying to tie P1.0 & P1.1 to ground and stopped messing around with the RST pin.  When you said "hundreds of milliseconds" it finally clicked in my head that the bootloader _was_ starting and it isn't anything like OptiBoot for AVRs or nearly as complicated as I had imagined.

This cut down on the variables which let me focus on trying all the baud rates until `-l 9600` finally worked.

grigorig, 2016-04-14:
It's odd that you actually had to change the handshake baudrate, though. Can you confirm that only 9600 works? What about 2400?

rgm3, 2016-04-15:
I rigged up an N-channel MOSFET to DTR so that I could do this:
```bash
#!/bin/bash

bauds=( 1200 2400 4800 9600 19200 38400 57600 115200 )

for (( i = 0; i < ${#bauds[@]}; i++ )); do
  for (( j = 0; j < ${#bauds[@]}; j++ )); do
    printf -- "stcgal -a -P stc12 -l %6d -b %6d : " ${bauds[j]} ${bauds[i]} >> results.txt
    stcgal -a -P stc12 -l ${bauds[j]} -b ${bauds[i]} main.hex
    stat=$?
    if [ $stat -ne 0 ]; then
      echo "Fail ${stat}" >> results.txt
    else
      echo "Pass" >> results.txt
    fi
  done
done
```
Here are the results.  All the Fail 1s are "Protocol error: incorrect frame start" I believe.  The other status is me hitting CTRL-C when stuck "waiting."
```
stcgal -a -P stc12 -l   1200 -b   1200 : Fail 2
stcgal -a -P stc12 -l   2400 -b   1200 : Fail 1
stcgal -a -P stc12 -l   4800 -b   1200 : Fail 1
stcgal -a -P stc12 -l   9600 -b   1200 : Fail 1
stcgal -a -P stc12 -l  19200 -b   1200 : Fail 1
stcgal -a -P stc12 -l  38400 -b   1200 : Fail 1
stcgal -a -P stc12 -l  57600 -b   1200 : Fail 1
stcgal -a -P stc12 -l 115200 -b   1200 : Fail 1
stcgal -a -P stc12 -l   1200 -b   2400 : Fail 2
stcgal -a -P stc12 -l   2400 -b   2400 : Fail 1
stcgal -a -P stc12 -l   4800 -b   2400 : Fail 1
stcgal -a -P stc12 -l   9600 -b   2400 : Fail 1
stcgal -a -P stc12 -l  19200 -b   2400 : Fail 1
stcgal -a -P stc12 -l  38400 -b   2400 : Fail 1
stcgal -a -P stc12 -l  57600 -b   2400 : Fail 1
stcgal -a -P stc12 -l 115200 -b   2400 : Fail 1
stcgal -a -P stc12 -l   1200 -b   4800 : Fail 2
stcgal -a -P stc12 -l   2400 -b   4800 : Pass
stcgal -a -P stc12 -l   4800 -b   4800 : Pass
stcgal -a -P stc12 -l   9600 -b   4800 : Pass
stcgal -a -P stc12 -l  19200 -b   4800 : Pass
stcgal -a -P stc12 -l  38400 -b   4800 : Pass
stcgal -a -P stc12 -l  57600 -b   4800 : Fail 1
stcgal -a -P stc12 -l 115200 -b   4800 : Fail 1
stcgal -a -P stc12 -l   1200 -b   9600 : Fail 2
stcgal -a -P stc12 -l   2400 -b   9600 : Pass
stcgal -a -P stc12 -l   4800 -b   9600 : Pass
stcgal -a -P stc12 -l   9600 -b   9600 : Pass
stcgal -a -P stc12 -l  19200 -b   9600 : Pass
stcgal -a -P stc12 -l  38400 -b   9600 : Pass
stcgal -a -P stc12 -l  57600 -b   9600 : Fail 1
stcgal -a -P stc12 -l 115200 -b   9600 : Fail 1
stcgal -a -P stc12 -l   1200 -b  19200 : Fail 2
stcgal -a -P stc12 -l   2400 -b  19200 : Pass
stcgal -a -P stc12 -l   4800 -b  19200 : Pass
stcgal -a -P stc12 -l   9600 -b  19200 : Pass
stcgal -a -P stc12 -l  19200 -b  19200 : Pass
stcgal -a -P stc12 -l  38400 -b  19200 : Pass
stcgal -a -P stc12 -l  57600 -b  19200 : Fail 1
stcgal -a -P stc12 -l 115200 -b  19200 : Fail 1
stcgal -a -P stc12 -l   1200 -b  38400 : Fail 2
stcgal -a -P stc12 -l   2400 -b  38400 : Pass
stcgal -a -P stc12 -l   4800 -b  38400 : Pass
stcgal -a -P stc12 -l   9600 -b  38400 : Pass
stcgal -a -P stc12 -l  19200 -b  38400 : Pass
stcgal -a -P stc12 -l  38400 -b  38400 : Pass
stcgal -a -P stc12 -l  57600 -b  38400 : Fail 1
stcgal -a -P stc12 -l 115200 -b  38400 : Fail 1
stcgal -a -P stc12 -l   1200 -b  57600 : Fail 2
stcgal -a -P stc12 -l   2400 -b  57600 : Pass
stcgal -a -P stc12 -l   4800 -b  57600 : Pass
stcgal -a -P stc12 -l   9600 -b  57600 : Pass
stcgal -a -P stc12 -l  19200 -b  57600 : Pass
stcgal -a -P stc12 -l  38400 -b  57600 : Pass
stcgal -a -P stc12 -l  57600 -b  57600 : Pass
stcgal -a -P stc12 -l 115200 -b  57600 : Fail 1
stcgal -a -P stc12 -l   1200 -b 115200 : Fail 2
stcgal -a -P stc12 -l   2400 -b 115200 : Pass
stcgal -a -P stc12 -l   4800 -b 115200 : Pass
stcgal -a -P stc12 -l   9600 -b 115200 : Pass
stcgal -a -P stc12 -l  19200 -b 115200 : Pass
stcgal -a -P stc12 -l  38400 -b 115200 : Pass
stcgal -a -P stc12 -l  57600 -b 115200 : Fail 1
stcgal -a -P stc12 -l 115200 -b 115200 : Pass
```

grigorig, 2016-04-15:
That is a great analysis, thank you. It seems odd that 1200 is failing, it works great in all cases for me. Low handshake baudrates are beneficial for the timing calculations done to measure the operating frequency of the MCU and for the frequency calibration on STC 15 series. That's why I chose 1200.

Possibly there are some issues with FTDI UARTs, I guess. I'll do some testing with an FT232RL later. For the time being, I'll raise the handshake baudrate to 2400.

denniswittich, 2019-10-12:
Hey all,
it's 3.5 years later and people are still struggeling with the same problems (at least me) haha....
However, my problem is a little different.. 

I also want to connect to a STC12C5A60S2. When I run
 **stcgal -P stc12 -p /dev/tty -a -l 9600**
I get **Unexpected error: (22, 'Invalid argument')**

I would be very thankful to get any hints or thoughts, since I have no more ideas. Is it somehow possible, that the microcontroller is not working at all? Is there any way to check?