grigorig/stcgal #14: Setting options for some STC12 series MCUs

laborer, 2016-05-17:
https://github.com/grigorig/stcgal/blob/master/stcgal/protocols.py#L1449

The last msr byte, `msr[3]`, is located at a different place in the option setting packet for STC12C5A16S2, STC10F104XE and STC11F02E.  So for these MCUs, `packet` should be
```python
        packet = bytes([0x8d, msr[0], msr[1], msr[2], 0xff,
                        0xff, 0xff, 0xff, 0xff, msr[3], 0xff,
                        0xff, 0xff, 0xff, 0xff, 0xff, 0xff])
```
And in the status packet, `msr[3]` is `status_packet[28]` rather than `status_packet[27]` as used [here](https://github.com/grigorig/stcgal/blob/master/stcgal/protocols.py#L1340).

I also tested STC12C5608AD.  Its status packet and option setting packet have different formats than the MCUs above and the formats currently used in _stcgal_.

grigorig, 2016-05-17:
Thank you for reporting. There are issues with some STC12A chips, too. That should include the STC12C5608AD, IIRC. I guess I'll do another round of verification and comparison against the official flash tool.

The real problem is of course, why is this stuff so damn messy! All of STC's bootloader protocols are just horrible.

laborer, 2016-05-17:
STC really needs to get its s*** together.  Sadly, it won't happen though.  STC is pretty much a one-man company, whose CEO has his cellphone number printed on datasheet...

Should STC12C5608AD be categorized as STC12A?  I assigned it to STC12 in automatic protocol detection.  I need to check the code and protocol.

grigorig, 2016-05-17:
I'm not actually sure if the STC12C5608AD should use the STC12A protocol. If it is wrongly categorized, it clearly shouldn't work at all (parity vs no parity).

laborer, 2016-05-17:
Yes, STC12C5608AD should be STC12 as it uses parity.  But since STC12C5608AD and STC12C5204AD use a different option format than other STC12s, maybe there needs to be a STC12B...

grigorig, 2016-05-17:
The STC12C5608AD definitely looks like a weird "hybrid" between STC12 and STC12A. In STC-ISP, the settings panel looks just like the other STC12A parts. I can't test anything though, I don't have any of these MCUs. If you figure out what's going on, that would be great.

I have some of the other STC12 parts so I'll look what is going on with these.

laborer, 2016-05-17:
The option packets of STC12C2052/STC12C5608AD/STC12C5204AD look almost identical.  Their BSL versions are very similar as well (5.8D/6.0G/6.0H, respectively).

grigorig, 2016-05-17:
Just for some quick comparisons, can you post a dump of a status and option set packet for the STC12C5608AD?

laborer, 2016-05-17:
STC12C2052

```
status
 46 B9 68 00 28 00 05 64 05 65 05 65 05 64 05 65
 05 65 05 65 05 64 58 44 00 F2 02 83 FD F7 F7 FF
 FF FF BF FF FD F7 F7 FF 80 16

set option
 46 B9 6A 00 26 8D FD F7 F7 FF BF 00 56 B2 A4 BF
 FF FD F7 FF FF FF FF F7 FF FF FF FF FF FF FF 00
 56 B2 A4 FF FF FF B9 16
```

STC12C5608AD
```
status
 46 B9 68 00 31 50 05 0C 05 15 05 0C 05 16 05 0C
 05 16 05 0D 05 16 60 47 00 E6 68 8E FD F7 F7 FF
 FF FF BF FF FD F7 F7 FF 00 00 00 00 00 00 00 00
 0F AC 16

set option
 46 B9 6A 00 27 8D FD F7 F7 FF FF 00 51 AC ED BF
 FF FD F7 FF FF FF FF F7 FF FF FF FF FF FF FF 00
 51 AC ED FF FF FF 1C 76 16
```

STC12C5204AD
```
status
 46 B9 68 00 31 50 09 C3 09 AF 09 C1 09 AF 09 C1
 09 AF 09 C0 09 AD 60 48 00 E1 64 8C 7D F7 F7 FF
 FF FF FF FF 7D F7 F7 FF 00 00 00 00 00 00 00 00
 14 39 16

set option
 46 B9 6A 00 27 8D 7D F7 F7 FF FF 00 9B A1 12 FF
 FF 7D F7 FF FF FF FF F7 FF FF FF FF FF FF FF 00
 9B A1 12 FF FF FF 1A 7E 16
```

grigorig, 2016-05-18:
STC12 option handling may have been wrong from the very start. Well, it's fixed now.  Let's keep this bug open until the "STC12B" variant has been figured out. The option packets are definitely exactly the same, just the framing is different.

grigorig, 2016-05-18:
There are still some open questions regarding STC12B. Can you provide completely dumps for one of these MCUs?

laborer, 2016-05-18:
The option packets seem to only differ in that `0xBF` byte.  I haven't figured out what that is for, but it might not be important.

laborer, 2016-05-18:
Here's a complete serial port log of programming STC12C5608AD

```
Port opened by process "stc-isp-15xx-v6.85N.exe" (PID: 832)

Request: 2016-5-18 9:58:20.99064
 7F 7F 7F 7F 7F 7F 7F 7F 7F 7F                           

Answer: 2016-5-18 9:58:21.36264 (+0.0000 seconds)
 46 B9 68 00 31 50 05 14 05 1E 05 14 05 1D 05 13   F¹h.1P..........
 05 1E 05 13 05 1D 60 47 00 E6 68 8E FD D7 F5 FF   ......`G.æhý×õÿ
 FF FF BF FF FD D7 F5 FF 00 00 00 00 00 00 00 00   ÿÿ¿ÿý×õÿ........
 0F A4 16                                          .¤.             

Request: 2016-5-18 9:58:22.61364 (+0.0200 seconds)
 46 B9 6A 00 0D 50 00 00 36 01 E6 68 02 4C 16      F¹j..P..6.æh.L. 

Answer: 2016-5-18 9:58:22.68364 (+0.0701 seconds)
 46 B9 68 00 07 8F 00 FE 16                        F¹h...þ.       

Request: 2016-5-18 9:58:22.73364 (+0.0100 seconds)
 46 B9 6A 00 0D 8F C0 E9 3F 2E 28 84 03 C8 16      F¹j..Àé?.(.È. 

Answer: 2016-5-18 9:58:22.03364 (+0.3004 seconds)
 46 B9 68 00 0E 8F C0 E9 3F 2E 28 84 03 03 CA 16   F¹h..Àé?.(..Ê.

Request: 2016-5-18 9:58:22.12364 (+0.0701 seconds)
 46 B9 6A 00 0C 8E C0 E9 3F 2E 28 03 42 16         F¹j..Àé?.(.B.  

Answer: 2016-5-18 9:58:22.42464 (+0.3004 seconds)
 46 B9 68 00 0D 84 C0 E9 3F 2E 28 03 03 3A 16      F¹h..Àé?.(..:. 

Request: 2016-5-18 9:58:22.45464 (+0.0200 seconds)
 46 B9 6A 00 8C 84 00 00 02 00 00 20 00 00 00 00   F¹j...... ....
 00 00 00 00 00 00 00 00 80 7F 7E 7D 7C 7B 7A 79   ........~}|{zy
 78 77 76 75 74 73 72 71 70 6F 6E 6D 6C 6B 6A 69   xwvutsrqponmlkji
 68 67 66 65 64 63 62 61 60 5F 5E 5D 5C 5B 5A 59   hgfedcba`_^]\[ZY
 58 57 56 55 54 53 52 51 50 4F 4E 4D 4C 4B 4A 49   XWVUTSRQPONMLKJI
 48 47 46 45 44 43 42 41 40 3F 3E 3D 3C 3B 3A 39   HGFEDCBA@?>=<;:9
 38 37 36 35 34 33 32 31 30 2F 2E 2D 2C 2B 2A 29   876543210/.-,+*)
 28 27 26 25 24 23 22 21 20 1F 1E 1D 1C 1B 1A 19   ('&%$#"! .......
 18 17 16 15 14 13 12 11 10 0F 0E 21 81 16         ...........!.  

Answer: 2016-5-18 9:58:27.69064 (+4.2361 seconds)
 46 B9 68 00 07 00 00 6F 16                        F¹h....o.       

Request: 2016-5-18 9:58:27.73064 (+0.0300 seconds)
 46 B9 6A 00 8D 00 00 00 00 00 00 80 AA AA AA AA   F¹j.......ªªªª
 AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA   ªªªªªªªªªªªªªªªª
 AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA   ªªªªªªªªªªªªªªªª
 AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA   ªªªªªªªªªªªªªªªª
 AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA   ªªªªªªªªªªªªªªªª
 AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA   ªªªªªªªªªªªªªªªª
 AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA   ªªªªªªªªªªªªªªªª
 AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA   ªªªªªªªªªªªªªªªª
 AA AA AA AA AA AA AA AA AA AA AA AA 56 77 16      ªªªªªªªªªªªªVw. 

Answer: 2016-5-18 9:58:27.84064 (+0.1102 seconds)
 46 B9 68 00 08 00 00 00 70 16                     F¹h.....p.      

Request: 2016-5-18 9:58:27.87064 (+0.0200 seconds)
 46 B9 6A 00 8D 00 00 00 00 80 00 80 AA AA AA AA   F¹j......ªªªª
 AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA   ªªªªªªªªªªªªªªªª
 AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA   ªªªªªªªªªªªªªªªª
 AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA   ªªªªªªªªªªªªªªªª
 AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA   ªªªªªªªªªªªªªªªª
 AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA   ªªªªªªªªªªªªªªªª
 AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA   ªªªªªªªªªªªªªªªª
 AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA   ªªªªªªªªªªªªªªªª
 AA AA AA AA AA AA AA AA AA AA AA AA 56 F7 16      ªªªªªªªªªªªªV÷. 
[... 27 lines cut ...]
 FF FF FF FF FF FF FF FF FF FF FF FF 81 78 16      ÿÿÿÿÿÿÿÿÿÿÿÿx. 

Answer: 2016-5-18 9:58:27.25164 (+0.1102 seconds)
 46 B9 68 00 08 00 80 00 F0 16                     F¹h....ð.      

Request: 2016-5-18 9:58:27.26164 (+0.0000 seconds)
 46 B9 6A 00 0D 69 00 00 36 01 E6 68 02 65 16      F¹j..i..6.æh.e. 

Answer: 2016-5-18 9:58:27.35164 (+0.0901 seconds)
 46 B9 68 00 07 8D 00 FC 16                        F¹h...ü.       

Request: 2016-5-18 9:58:27.37164 (+0.0200 seconds)
 46 B9 6A 00 27 8D FD F7 F7 FF FF 00 52 3D 92 BF   F¹j.'ý÷÷ÿÿ.R=¿
 FF FD F7 FF FF FF FF F7 FF FF FF FF FF FF FF 00   ÿý÷ÿÿÿÿ÷ÿÿÿÿÿÿÿ.
 52 3D 92 FF FF FF 1A E4 16                        R=ÿÿÿ.ä.       

Answer: 2016-5-18 9:58:27.50164 (+0.1302 seconds)
 46 B9 68 00 1C 50 FD F7 F7 FF FF 03 FF 60 47 FD   F¹h..Pý÷÷ÿÿ.ÿ`Gý
 F7 F7 FF FF FF BF FF FD F7 F7 FF 12 F6 16         ÷÷ÿÿÿ¿ÿý÷÷ÿ.ö.  

Request: 2016-5-18 9:58:27.54164 (+0.0200 seconds)
 46 B9 6A 00 07 82 00 F3 16                        F¹j...ó.       

Port closed
```

grigorig, 2016-05-18:
The 0xBF byte in the set option packet most likely is the MCS3 byte as described in https://github.com/grigorig/stcgal/blob/master/doc/stc12a-options.txt, so it configures the low voltage reset level. Maybe you could check that. Thanks for the full dump. Should be good enough to implement STC12B "blind" for now.

laborer, 2016-05-18:
That `0xBF` is indeed the LVD reset level.  There are two `0xBF` bytes in an option packet of STC12C2052, but it appears that only the second one, which is also the byte used in STC12C5608, is effective.  So these STC12Bs should be able to share the same option packet.

grigorig, 2016-05-19:
https://github.com/grigorig/stcgal/tree/stc12b has some preliminary support, still thinking of a saner architecturer. :)

laborer, 2016-05-19:
I like this solution.  The same idea can help improve the inheritance structure.

By the way, I think it is better to not send the option packet when user doesn't request an change of an option in command line.  Just to be safe.

grigorig, 2016-05-20:
> I think it is better to not send the option packet when user doesn't request an change of an option in command line.

Maybe. However, the option packet is also used to store the currently known operating frequency, trim values, etc. which complicates things. I'll leave it enabled always for now.

closed 2016-05-20 by grigorig