grigorig/stcgal #18: OT: kind help on STC microcontrollers dump fw

Barabba11, 2016-08-24:
Hi, thank you for your kind attention and sory for the off-topic. It's hard for me to find someone who has enogh experience on STC to clear an important issue, I hope you can help me.. thank you a lot!

I've a club light "LED bar" driven by DMX line, inside board there is an STC 12c5616ad and repairing the lamp I broken it. There is no service for this lamp available so I would like to clone one STC reading the firmware from another equal lamp (I have). Is that possible?

I downloaded the datasheet but they don't talk about firmware protection, neither how to dump it, if you never deal with such feature, do you know someone who has experience how to proceed?

I've another question, it looks programming these chips need only an RS232 port on a PC and some components to interface the RS232 line levels (to TTL), I got it right? Then it looks very simple to program/read it.

Thank you a lot for your kind support
Giovanni

grigorig, 2016-08-24:
> I downloaded the datasheet but they don't talk about firmware protection, neither how to dump it, if you never deal with such feature, do you know someone who has experience how to proceed?

There's a reason for that: there is no official way to dump the flash memory. Therefore there's no need for any kind of protection bits or the like, because everything is always protected. So, unfortunately, you are out of luck.

In issue #7 we've started to discuss possible loopholes, but so far, nothing has come up.

> it looks programming these chips need only an RS232 port on a PC and some components to interface the RS232 line levels (to TTL), I got it right? Then it looks very simple to program/read it.

Yes. Programming these chips is simple from a hardware POV.

Barabba11, 2016-08-24:
thank you for your answer. Ok I should consider it lost then.. really a pity

Barabba11, 2016-08-25:
Hi! yes, the Involight LED Bar 390, which has the same board of Involight LED Bar 400 and I suppose many other models.
I can forward you the electric scheme, which regarding the led drive (most important) is very simple. The light also has a display driven by shift registers, 2 buttons for changing between DMX, Audio and static programs (configurable).
It consist only in a TTL inverter used as buffer/STC protection, and 2 stadium (each is driving 12 leds, ahlf lamp) of 3 PT4115 (R,G,B, channels), so in total 6 PT4115. 
4115 is a very cheap led driver, capable to drive 3watt led up to 30 Volt, it just need few components, and in this scheme they used the recomended scheme on datasheet. Each 4115 drives 6+6 leds (in parallel).

Barabba11, 2016-08-25:
Actually I'm looking for a DMX "translator".. I mean.. that's an ambitious project but it can be revoluctionary in terms of lighting effects in club. I try to explain:
Lasers are becoming very popular, and cheap, but there isn't out yet a cheap solution to drive them "live".. Usually DMX software/consoles are made for it, but lasers supports by DMX a really reduced set of presettings comparing what they can do by ILDA interface. ILDA is used by laserdesign softwares, which produces .ild files and most of them can also drive live the laser.. but how to sincronize it with a DMX consoles? 
THere are different ways, some support MIDI for example, but it's need to output from a DMX software in MIDI (some support.. and some in very basic mode (only string output), or convert the DMX output to MIDI (looks output from a PC and enter in the same PC again, what a stupid thing), or.. and here is interesting for us.. play "wav files" (generated by .ILD) into USB audo cards that drive the ILDA interface.. 
To play wav files with low latency it is just needed a Raspberry (that can play unlimited wave files stored on usb flash/SD card),  but I suppose Raspberry can't even check a DMX line.. to don't lose the smooth ad accurate wav output timings, so a microcontroller that can sniff a DMX line and provide I2C meggages using interrupts only when needed (DMX address matched) could sound great. With DMX it's possible to trigger any .ild sequence, offest it X-Y, cahnce colors, speed and intensity, and so on; and place a little box with Raspberry close to laser, far from console, serving it only with DMX line and ethernet to upload new effects.)
It's just an idea, I don't know if u're interested to

Barabba11, 2016-09-05:
Hi, 
in case you're interested here is the board, I can forward you the electrical scheme. Actually it's very easy, there is the DMX input and 3 output (RGB), PWM, which drives a TTL buffer and then the PT4115 which does just need am inductance, a diode, a resistor and a capacitor. It supports wide range of power supply and can drive any diode till 3W.
http://i64.tinypic.com/24x2b0p.jpg

grigorig, 2016-09-21:
At this point the discussion isn't really specific to stcgal anymore. So let's close this. It probably makes sense to take further discussion elsewhere.

closed 2016-09-21 by grigorig