grigorig/stcgal #19: STC15W4K56S4 (BSL: 7.3.5T) fails to upload

tpimh, 2016-11-11:
Here is the output:

```
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 34 50 8F 0C 84 A3 F7 BC 9F 00 A8 B6 A0 FD 27 1E 00 00 73 54 00 F5 28 05 06 70 A3 02 10 14 19 1B 1E 00 EC E0 04 F3 EC 84 BF FF FF 16 07 14 60 13 3B 16
done
Protocol detected: stc15
Target model:
  Name: STC15W4K56S4
  Magic: F528
  Code flash: 56.0 KB
  EEPROM flash: 3.0 KB
Target frequency: 11.057 MHz
Target BSL version: 7.3.5T
Target wakeup frequency: 36.620 KHz
Target options:
  reset_pin_enabled=False
  clock_source=internal
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=4
  eeprom_lvd_inhibit=True
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart2_passthrough=False
  uart2_pin_mode=normal
  cpu_core_voltage=unknown
Loading flash: 162 bytes (Intel HEX)
Trimming frequency: -> Packet data: 46 B9 6A 00 20 00 0C 00 C0 80 C0 FF C0 00 80 80 80 FF 80 00 40 80 40 FF 40 00 00 80 00 C0 00 0A D3 16
Serial port error: pulse timeout
```

grigorig, 2016-11-11:
What kind of serial dongle do you use, i.e. what chipset does it have?

tpimh, 2016-11-11:
It uses CH340G. I tried different baudrates, but the result is always the same. I'm using this board:
[image]
[image]

grigorig, 2016-11-11:
Yeah, that chipset is buggy with Linux at the moment. Parity doesn't work, which is required for programming STC microcontrollers. Are you using Linux? There are some changes to fix this queued up for the Linux driver that are going to be included in Linux 4.10.

grigorig, 2016-11-11:
Of course, if you feel adventurous, you could try to cherry-pick the ch341 commits from this branch:

https://git.kernel.org/cgit/linux/kernel/git/johan/usb-serial.git/log/?h=usb-next

tpimh, 2016-11-11:
Yes, I'm on Linux. Probably, I'm going to get rid of this USB-Serial converter and connect USB lines directly to MCU, so I can use `usb15` mode.

tpimh, 2016-11-11:
Tried uploading with CP2102 and it works. So this issue is irrelevant to stcgal, the problem is in the kernel module.

closed 2016-11-11 by tpimh