grigorig/stcgal #21: unable to upload to STC15L204EA

edillmann, 2017-01-22:
Hi,

I just tried to query an STC15L204EA device: 

```
stcgal -D -t 11059 -P stc15a -l 2400 -b 9600 -p /dev/ttyUSB0
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 07 80 00 EF 16
-> Packet data: 46 B9 6A 00 07 80 00 F1 16
<- Packet data: 46 B9 68 00 07 80 00 EF 16
Protocol error: incorrect magic in status packet
-> Packet data: 46 B9 6A 00 07 82 00 F3 16
Disconnected!
```

without success, any clues ?

Regards,
Eric

grigorig, 2017-01-23:
Are you using one of the infamous CH240/CH341 USB UART dongles?

edillmann, 2017-01-23:
yes i'm using a CH341 dongle, so could you give me the reference of a working one ?

Thanks,
Eric

grigorig, 2017-01-23:
The Linux driver is unfortunately lacking parity support, which is required. This recently was merged into Linux 4.10. If you use Linux, either use another UART adapter or wait for the updated kernel.

edillmann, 2017-01-23:
i will give it a try, and report back,

thank's for your support

edillmann, 2017-01-23:
I confirm upgrading kernel to 4.10.0-rc5 did the trick, one more thanks ;-)

closed 2017-01-23 by edillmann

edillmann, 2017-01-23:
you can add STC15L204EA to compatibility report :-)