grigorig/stcgal #22: stc89 packet checksum mismatch

wapznw, 2017-03-13:
执行:
```
./stcgal.py -P auto ../c51demo/MAIN.hex
```
提示错误:
```
Waiting for MCU, please cycle power: done
Protocol detected: stc89
Protocol error: packet checksum mismatch
Disconnected!
```

执行：
```
./stcgal.py -P stc89 ../c51demo/MAIN.hex
```

提示：
```
Waiting for MCU, please cycle power:
```

开发板重新上电后没反应。

系统： OS X:10.10.5

grigorig, 2017-03-13:
Can you provide any more details about the board and MCU model? Also, please write in English, if possible.

grigorig, 2017-07-03:
This might be a duplicate of #8.

Destinux, 2017-08-02:
This issue perhaps has same problem as my issue.
there just some chinese like
run:
raise error:
run:
stay the cycle waiting:
restart develope board and have no reaction.
System is OSX 10

SUCHMOKUO, 2018-11-14:
Same issue for me..

falalal, 2018-11-27:
LwwdeMacBook-Air:Desktop lww$ stcgal -a -P stc89  -p /dev/tty.wchusbserial1410 -b 9600 -l 9600 -D  main.ihx 
Cycling power: done
Waiting for MCU: <- Packet data: 00
<- Packet data: 46 B9 68 00 29 50 FD 00 00 00 00 00 00 00 00 00 00 00 00 5F 00 00 72 43 00 F0 51 05 80 00 FF FF FF FF 38 20 18 09 06 60 09 93 16
<- Packet data: 00
<- Packet data: 00
<- Packet data: 46 B9 68 00 29 50 FD 00 00 00 00 00 00 00 00 00 00 00 00 5F 00 00 72 43 00 F0 51 05 80 00 FF FF FF FF 38 20 18 09 06 60 09 93 16
<- Packet data: 30
<- Packet data: 2E
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 56
<- Packet data: 20
<- Packet data: 20
<- Packet data: 4E
<- Packet data: 61
<- Packet data: 4E
<- Packet data: 63
<- Packet data: 6D
<- Packet data: 0D
<- Packet data: 0A
<- Packet data: 30
<- Packet data: 2E
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 56
<- Packet data: 20
<- Packet data: 20
<- Packet data: 4E
<- Packet data: 61
<- Packet data: 4E
<- Packet data: 63
<- Packet data: 6D
<- Packet data: 0D
<- Packet data: 0A
<- Packet data: 30
<- Packet data: 2E
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 56
<- Packet data: 20
<- Packet data: 20
<- Packet data: 4E
<- Packet data: 61
<- Packet data: 4E
<- Packet data: 63
<- Packet data: 6D
<- Packet data: 0D
<- Packet data: 0A
<- Packet data: 30
<- Packet data: 2E
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
It just cannot stop.What's the problem~help

falalal, 2018-11-27:
By the way, I have the same problem.

ConnorFi, 2019-01-19:
Ubuntu 16.04 STC89C52 by the way

histriver, 2019-07-20:
As luck i have two STC MCU, one is STC89C52, another is STC90C516RD+. the experiment show that STC90C516RD+ is all right and can write success,  the STC89C52 show the same error: Protocol error: packet checksum mismatch.  the os is macOS High Sierra 10.13.6.  

so i think i need buy another STC89C52RC to replace STC89C52. maybe it will work normal.

TomXia, 2019-08-01:
Hi there. @grigorig @ Thanks for your efforts.
Same issue for me.
`localhost:~ tomxia$ stcgal -p /dev/tty.wchusbserial1410 -D
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 29 50 FD 01 87 43 49 8A 0B C3 6A 84 2B 0B 01 9F A9 F1 72 43 00 F0 51 05 80 00 FF FF FF FF 38 20 19 04 08 60 0E FC 16
done
Protocol detected: stc89
<- Packet data: 46 B9 68 00 29 50 FD 01 87 43 49 8A 0B C3 6A 84 2B 0B 01 9F A9 F1 72 43 00 F0 51 05 80 00 FF FF FF FF 38 20 19 04 08 60 0E FC 16
Protocol error: packet checksum mismatch
-> Packet data: 46 B9 6A 00 06 82 F2 16
Disconnected!
`
CH340C, STC89C52, macOS 10.13.6
In my case, stcgal is not working on Windows as well with same error.
The MCU works well while using STC-ISP on windows VM.

yuangu, 2020-02-25:
Same issue for me.
```
 stcgal -P auto  -p /dev/tty.wchusbserial14320 -D
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 29 50 FD 20 20 48 65 6C 6C 6F 20 57 6F 72 01 7F 21 20 72 43 00 F0 51 05 80 00 FF FF FF FF 38 20 19 09 09 60 0D 85 16
done
Protocol detected: stc89
<- Packet data: 46 B9 68 00 29 50 FD 20 20 48 65 6C 6C 6F 20 57 6F 72 01 7F 21 20 72 43 00 F0 51 05 80 00 FF FF FF FF 38 20 19 09 09 60 0D 85 16
Protocol error: packet checksum mismatch
-> Packet data: 46 B9 6A 00 06 82 F2 16
Disconnected!
```

homanlokqwq, 2023-04-02:
broad 8051
os ubuntu20
same

grigorig, 2023-06-02:
If any one of you is still here: newer STC89 devices have a newer bootloader with some protocol changes. Please try again and specify the "stc89a" protocol manually.

closed 2023-06-02 by grigorig