grigorig/stcgal #23: Anyone tried STC8?

tpimh, 2017-05-07:
I wanted to buy this board with STC8A8K64S4A12:
[image]
I wonder if anyone tried programming it with stcgal. What are the chances that I will work with `stc15` or `usb15` protocol?

Here is the datasheet: http://www.stcisp.com/datasheet/stc/STC8F-datasheet.pdf (in Chinese)

grigorig, 2017-05-07:
This is a new series, so I'm quite sure there are incompatibilities.

grigorig, 2017-07-19:
FYI STC8 should be in progress. As soon as my board arrives.

xcvista, 2017-08-16:
@grigorig Make sure your board is carrying the final version of the chip though. I got one based on the prerelease "C" version but the release version is version "F" for the STC8A8K variants and "D" version for STC8F2K variants. The board above is a version "C" chip, as shown in the last letter on the markings.

grigorig, 2017-08-17:
Well, apparently my version is "E". Hopefully that's good enough.

tpimh, 2017-08-17:
I think support for all existing versions should be implemented. I ordered some STC8A8K64S4A12, don't yet know which revision I will get.

grigorig, 2017-08-19:
Of course! Ideally, all revisions should work. STC often doesn't make it easy with their chaotic versioning, though.

tpimh, 2017-08-19:
I hope these C/D/E/F revisions are not very different from each other, so that it is not too much work to support all of them. But first we should at least get all of the existing versions, having contacts in STC would be ideal.

xcvista, 2017-08-19:
We should focus our efforts on the final release chips, and only after the final release chips are properly supported should we start trying the code on prerelease.

I live close to the company. Maybe I can ask them for samples of the chips? They may offer only the final release version though.

iglooom, 2018-01-07:
Any news about STC8 support?

grigorig, 2018-01-09:
Try https://github.com/grigorig/stcgal/tree/stc8

I haven't had much free time to finish it. Flashing seems to work fine, but it's not possible to set options yet. Still, might be usable.

gh-jmg, 2018-04-06:
This looks great. Any updates on STC8 ?  (STC8F/STC8A/STC8H)
I see they move to general release, with https://lcsc.com/search?q=STc8f showing 20 items. 

Is this work doe UART loading, or is the STC8 USB loader also being done ? 
( I see mention of STC15W4  usb15 protocol, & I guess STC8 is very similar/identical ?)

grigorig, 2018-04-07:
I haven't had a chance to look at the USB protocol yet. But it's probably similar an should be much easier to implement than UART based protocol. (No frequency calibration or any other "fun" things like that)

I might mainline STC8 UART support without support for options in the meantime. It's already useful as-is.

hkoutram, 2018-07-24:
Anybody have success programming the STC8F2K64S4-28I-DIP40 chip. I just got ten of them in the mail.  Does the STC15 type interface work? I have the 6.86F version S/W.

gh-jmg, 2018-07-24:
I notice here
https://github.com/grigorig/stcgal/blob/stc8/stcgal/protocols.py
changes made 28 days ago, so it seems STC8 is progressing.

Comments seem around uncalibrated devices, which is puzzling, - I think all
full release versions of STC8 are calibrated  - maybe that is for either
samples (not calibrated yet) or bricked parts, where CAL was damaged ?

Those comments also suggest the STC8 have both USB and UART loaders ?  -
USB on the website and UART in this code ?

grigorig, 2018-07-24:
Yes, I'm working on STC8, on and off. Basic flashing should already work with the `stc8`branch.

grigorig, 2018-08-21:
Full-featured STC8 support (UART protocol) was finally merged into master and works flawlessly for me with STC8A8K64S4A12. Please open specific issues if you have trouble with it.

closed 2018-08-21 by grigorig