grigorig/stcgal #25: I am a new but just cant understand...

Destinux, 2017-08-02:
install and run `stcgal` for info, then...
```
Waiting for MCU, please cycle power: <- Packet data: 00
<- Packet data: 46 B9 68 00 29 50 FD 00 00 00 00 00 00 00 00 00 00 00 01 81 00 00 72 43 00 F0 51 05 80 00 FF FF FF FF 38 20 17 06 23 60 09 CF 16
done
Protocol detected: stc89
<- Packet data: 46 B9 68 00 29 50 FD 00 00 00 00 00 00 00 00 00 00 00 01 81 00 00 72 43 00 F0 51 05 80 00 FF FF FF FF 38 20 17 06 23 60 09 CF 16
Protocol error: packet checksum mismatch
-> Packet data: 46 B9 6A 00 06 82 F2 16
Disconnected!
```
why...

grigorig, 2017-08-02:
That looks a bit odd. It looks like stcgal is receiving the info packet twice. I'll check what's up.

Can you provide more info about your setup? I.e. which chip are you using, how are you connecting to your PC / which USB-UART are you using, what's your command line, which stcgal version, etc.

Destinux, 2017-08-02:
I use STC89C52, not STC89C52RC or another 89 series;
the UART chip is WCH CH340G;
My OS is Deepin(derive from Debian),and I just know use the default terminal;
I download stcgal from github just a few hours ago...how can I found version?

grigorig, 2017-09-15:
Does it change anything if you explicitly specify the protocol with `-P stc89`?

falalal, 2018-11-27:
LwwdeMacBook-Air:Desktop lww$ stcgal -a -P stc89  -p /dev/tty.wchusbserial1410 -b 9600 -l 9600 -D  main.ihx 
Cycling power: done
Waiting for MCU: <- Packet data: 00
<- Packet data: 46 B9 68 00 29 50 FD 00 00 00 00 00 00 00 00 00 00 00 00 5F 00 00 72 43 00 F0 51 05 80 00 FF FF FF FF 38 20 18 09 06 60 09 93 16
<- Packet data: 00
<- Packet data: 00
<- Packet data: 46 B9 68 00 29 50 FD 00 00 00 00 00 00 00 00 00 00 00 00 5F 00 00 72 43 00 F0 51 05 80 00 FF FF FF FF 38 20 18 09 06 60 09 93 16
<- Packet data: 30
<- Packet data: 2E
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 56
<- Packet data: 20
<- Packet data: 20
<- Packet data: 4E
<- Packet data: 61
<- Packet data: 4E
<- Packet data: 63
<- Packet data: 6D
<- Packet data: 0D
<- Packet data: 0A
<- Packet data: 30
<- Packet data: 2E
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 56
<- Packet data: 20
<- Packet data: 20
<- Packet data: 4E
<- Packet data: 61
<- Packet data: 4E
<- Packet data: 63
<- Packet data: 6D
<- Packet data: 0D
<- Packet data: 0A
<- Packet data: 30
<- Packet data: 2E
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 56
<- Packet data: 20
<- Packet data: 20
<- Packet data: 4E
<- Packet data: 61
<- Packet data: 4E
<- Packet data: 63
<- Packet data: 6D
<- Packet data: 0D
<- Packet data: 0A
<- Packet data: 30
<- Packet data: 2E
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
<- Packet data: 30
It just cannot stop.Can someone help me?

liuup, 2020-10-03:
Hello, bro, Is that problem be solved? I meet the same problem in my project, and I just want to know how can u solve it :) Hope for ur answer :)

grigorig, 2021-01-09:
This looks like a power supply problem.

closed 2021-01-09 by grigorig