grigorig/stcgal #30: Board recommendation?

larsbrinkhoff, 2017-10-13:
Hello,

Can you recommend an STC MCU development board, which is easy to program using stcgal?  I'd prefer a USB interface if possible.

I searched for "STC15W4" on eBay.  There were some interesting results, but the boards are equipped with ESP8266 parts, so I'm not sure if they're appropriate for 8051 programming.

Thanks.

nekromant, 2017-10-13:
ESP8266 are NOT STC, they have a different MCU and are NOT covered by any of these tools.

larsbrinkhoff, 2017-10-13:
I understand that.  This is really just tangential to the request for advice.  But I looked it up in more detail.  The boards are called "DC 5V IAP15W4K58S4 STC15W4K56S4 SCM DC5V ESP8266 Socket Development Board".  The IAP15W4K58S4 is an STC part number for an 8051 device.  It seems there is an STC MCU on board, but I'm not sure what its purpose is.

Anyway, I'm just looking for advice on a nice STC board.

zerog2k, 2017-10-13:
Lars, I have seen several of these boards, which combine some sort of STC host processor, along with an ESP8266 module - however the ESP is usually supplied with AT firmware, and all of the board peripherals (relay, leds, DHT, etc) are wired up to the host processor, the STC. In these cases, generally, the ESP is just for "wifi" access. Sadly, many of these boards are not well documented, e.g. schematics - at least not in English.

Here is an older example:
https://esp8266hints.wordpress.com/2015/10/25/nuther-new-something-in-town-you-can-safely-ignore-it/ 
(This would need also a cheap usb to uart adapter).
Sadly, the author doesn't take any interest in using the STC onboard.

larsbrinkhoff, 2017-10-16:
Thank you @zerog2k.

Actually, I'm not familiary with any STC device.  I just want an 8051 which is easy to program through USB.

nekromant, 2017-10-17:
Hm... ESP8266 Black board T5 looks like just what I needed for a little project, just ordered one. ;)

zerog2k, 2017-10-17:
@nekromant fyi in case you were not aware, that blackboard t5 has already had the code reworked which makes a great starting point...
https://github.com/darrylb123/blackboard-STC15

nekromant, 2017-10-17:
@zerog2k thanks, that will prove useful. I think of using this board to add tcp and serial transport s to my embedded rpc library: github.com/nekromant/aura

larsbrinkhoff, 2017-10-29:
Thanks all!

closed 2017-10-29 by larsbrinkhoff