grigorig/stcgal #34: Can't get past 'Waiting for MCU, please cycle power'

MarqueIV, 2017-12-01:
Using an Adafruit FTDI Friend on macOS High Sierra and trying to flash one of those DIY clocks from BangGood/Amazon.  Mine is running the STC 15F204EA.

I have my FTDI Friend's Power and Ground connected to the clock's power and ground pins, and I have the clock's TX and RX pins connected to the FTDI Friend's RX and TX pins accordingly.

Stcgal seems to open the port properly, but it just sits on 'Waiting for MCU, please cycle power'.  No matter what I do... plugging and unplugging the connectors as a whole, just pulling power (which doesn't work as the MCU still gets power from the TX pin), repeatedly connecting/disconnecting them all as a group, holding down the side buttons on the clock... nothing.  I can't get it to flash.

Thinking it may be OSX, I also tried flashing from Linux and Windows (both in a virtual machine) but the same thing... waiting to reboot.

I read somewhere that I may have to put a resistor inline with the TX pin due to phantom power, but I can't find the article that had that.

Do you have any idea why this won't flash?

zerog2k, 2017-12-02:
If you don't want to hookup DTR pin to reset pin of STC mcu, you can try manually disconnect only ground and reconnect. As you observed, mcu can continue to be powered by tx line idling high (I'm guessing via the clamping diodes).

nekromant, 2017-12-02:
Try a different (non-ftdi) dongle or check this one under Linux and windows. Some serial drivers are terminally broken under macos, and ftdi may be well the case.

MarqueIV, 2017-12-03:
I've actually now tried several different things, all with no success on my Mac (late 2016 MBP running High Sierra.)  Thinking Adafruit's FTDI Friend may be the issue, I picked up a kit on Amazon that had three different USB-to-serial adapters; one based on the CP2102, a second based on the CH340 and a third based on the PL2303 chip.  That way, including Adafruit's (configured at their default of 5.0 power, but 3.3 signaling) I had four different ones to try.

I also yanked the chip (28-pin DIP, not the 20-pin variant) from the clock and plugged it directly into a breadboard, attaching only power (pin 12) to 5v on the adapter, ground (pin 14) to ground on the adapter, TX (pin 15) to RX on the adapter, and RX (pin 16) to TX on the adapter. Same thing. Waiting for MCU.

I've use a four-pin connector so I could yank out all at the same time. I've used separate wires so I could disconnect power or ground separately, but still nothing.

Somewhere I read about maybe having to add a resistor to the adapter TX (MCU RX) line, but they didn't say how much to add, and I'm not sure how to calculate what would be needed.

All of that said, can you think of anything else I can try?  Should I try all of the above in a Linux VM running on the Mac?  I'm concerned a VM may affect things negatively.  I may have a Raspberry Pi Zero W I can surface to try, but I'm not sure that's a good candidate (or maybe it's a perfect candidate.  That's what I mean by 'not sure'! :))

MarqueIV, 2017-12-03:
Zero2G, I missed your comment.  What do you mean 'reset pin on the MCU'?  The FTDI Friend specifically mentions being able to connect a DTR so perhaps that's something good to try.  Even so, isn't that something I could just jump myself right on the breadboard without even needing the DTR connector?

The other thing is am I supposed to do anything in terminal when seeing that message (i.e. press return, some other combination, etc.) while resetting?  I noticed while I can press return, it just jumps to the next line.  I don't get any actual feedback what I'm doing is actually doing anything, or even if it's needed.

Still, I can't imagine it's supposed to be this complicated. I have to think I'm missing something small. (like a resistor! Ba da bing!!  I'm here all week, folks!!)

MarqueIV, 2017-12-03:
Also, here's the data sheet I'm working off of...

[STC15F204EA MCU.pdf](https://github.com/grigorig/stcgal/files/1525076/STC15F204EA.MCU.pdf)

grigorig, 2017-12-03:
The bootloader is only invoked on power-on reset. Phantom powering through the data lines can definitely be a thing depending on the overall circuit design, so adding series resistors to the TX/RX lines is a good option. Try 1K or something like that. It might also be a good option to short-circuit VCC to GND on the MCU in the 'off' state (maybe with a resistor, for safety).

DTR can optionally be used as a control line to automate the power toggling (this still requires extra hardware).

MarqueIV, 2017-12-04:
I'll give the resistor thing a try later tonight when I'm back home.  Didn't quite follow the 'vcc-and in the off-state' comment.  If the MCU is off, I assume you mean no power, and if there's no power, not sure what shorting will do.  Or am I missing something?

Also, can you confirm that I should be able to flash the chip from a breadboard connecting only power, ground, TX and RX (the latter two with 1K resistors)?  Can you also confirm that I can use the power and ground from the USB Serial device, right?  I'm pretty sure they're 5V but I'll double check that too.

If the above is correct, what do you recommend to trigger the reset disconnecting vcc or gnd, and should I also temporarily disconnect TX and RX at the same time?

Any other pointers?  Again, I have to think this is something I'm doing wrong.  I just wish someone had a step-by-step for people like me who are pulling our hair out not knowing where in the chain the issue is.

grigorig, 2017-12-04:
> I'll give the resistor thing a try later tonight when I'm back home. Didn't quite follow the 'vcc-and in the off-state' comment. If the MCU is off, I assume you mean no power, and if there's no power, not sure what shorting will do. Or am I missing something?

If the MCU is being "phantom powered" through the data lines, short circuiting Vcc should keep it from actually turning on.

>  Can you also confirm that I can use the power and ground from the USB Serial device, right? I'm pretty sure they're 5V but I'll double check that too.

Works fine for me with various USB serial adapters. They should always be able to supply the 20 mA or so needed.

> Any other pointers? Again, I have to think this is something I'm doing wrong. I just wish someone had a step-by-step for people like me who are pulling our hair out not knowing where in the chain the issue is.

I think most people don't have any issues. I've considered putting up an FAQ though that might include various hints.

MarqueIV, 2017-12-04:
"If the MCU is being "phantom powered" through the data lines, short circuiting Vcc should keep it from actually turning on."

But wouldn't disconnecting ground essentially do the same thing? If there's no ground, there's no circuit.  Or are you saying it's using TX/RX as both power and ground, and short-circuiting vcc essentially stops that from happening?

Even so, I've been using a four-pin connector that I'm unplugging, then plugging in which disconnects/reconnects them all at once so that has to have killed the power, no?

Regardless, just to recap...

1. I *can* simply use the MCU in a breadboard, connecting only VCC, CGND, TX and RX (correct?)
2. I should put 1K resistors in series on the TX/RX lines.
3. With power/ground disconnected I should also short out vcc to ground.
4. All I should have to do is run 'make flash' (after 'make clean' and 'make' of course)

LMK if I've missed anything, or if you can offer any additional information.  I know it may not seem like it, but there are actually a lot of us who get that 'waiting' error, at least according to Google anyway.

grigorig, 2017-12-04:
Everything usually works just fine for me without 2 and 3, with many different STC MCUs, different USB serial adapters, on Linux and Windows, VM or bare metal. I don't understand why you have so many issues, sorry.

And yes, also disconnecting ground may help to solve these "phantom power" issues as well.

MarqueIV, 2017-12-04:
Yeah, I don't know why either.  Nor do the others posting the same issues.  That's the confusing part.  Who knows... maybe it's a bad chip.  I've ordered another clock so I'll be trying that one too.

So... since we've (i.e. people in my predicament) have never seen anything showing what it *should* look like when you go through the steps, can you explain the exact ones you take, along with the 'when's' and what you see as output along the way?

For me, I run 'make flash', then get the 'waiting' message, and from there I disconnect and reconnect the four-pin wire connected to the breadboard.  The message of course never changes for me, but it does for you.  What comes next? Do you have to do anything else/press anything else?  And how long does it take to flash?  What are the command-line params you're using with stcgal?  Anything that can help would be appreciated.

zerog2k, 2017-12-05:
stcgal is just an opensource flash programming tool replacement for the windows-only, proprietary STC-ISP software.
If you are unable to program your chip, then you may have some circuit implementation issue.
Yes, should should be able to try to breadboard the barebones chip and see if you can program it there (refer to the datasheet for the programming minimum application circuit)

Since this is not an stcgal-specific issue, let's continue the thread on the issue you created on my stc_diyclock project (which just uses stcgal as preferred programming tool when you do `make flash`)
https://github.com/zerog2k/stc_diyclock/issues/27

MarqueIV, 2017-12-05:
We can continue there, but technically this *is* a STCGAL issue (or more accurately, an issue *I* am having with STCGAL, which itself may very well be fine) since that's the tool that's hanging regardless of what I'm trying to flash with it.

That said, I only found out about this because of your clock firmware so sure... we can continue there.  Honestly, I really don't care where it gets resolved so long as it does.

MarqueIV, 2017-12-09:
Ok, this is kind of weird!  I can now get this to flash simply by reversing the TX/RX lines.  Don't know why I decided to try it, but that worked.  What's confusing is this means TX on my USB-to-Serial adapter is connected to TX on the clock (and RX to RX) which, unless I'm mistaken, is backwards.  TX from adapter should go to RX on the clock.  But that doesn't work.

That said, now the clock is displaying something weird, and is just buzzing, so looks like I have another problem.  Still, at least this one is addressed as I am now flashing, so I'm going to close it out and continue over there.

closed 2017-12-09 by MarqueIV

falalal, 2018-11-27:
Waiting for MCU, please cycle power:it can't work no matter how many times I push the button to start.Can someone help me?

falalal, 2018-11-27:
I don't know what's the problem.It's STL 89c52.

venkat602609, 2022-10-03:
I have an STC12C5A60S2. The problem seems to be that the board is drawing power from the 5V UART Pin. I disconnected the 5V pin and cycled power and things worked. One other thing is that if the 5V power is not connected during programming, I would get an error message. So first disconnect the 5V pin from UART during discovery and connect it back during flashing.

area-8051, 2022-10-03:
I often get this message with other MCU too. I noticed that if I start STCGAL while the previous version of the firmware is doing a lot of I/O, the upload always succeeds. If the old firmware is in its super loop doing nothing (i.e. a tight loop), the upload may fail, often repeatedly.