grigorig/stcgal #35: AttributeError: module 'serial' has no attribute 'PARITY_NONE'

Zontex, 2017-12-11:
Hi,
I'm working on Mac OSX And I've installed python3 and pip3 and installed the stcgal package using sudo python3 setup.py install but for some reason when I run the software I get the following error:
`
stcgal -h
Traceback (most recent call last):
  File "/usr/local/bin/stcgal", line 11, in <module>
    load_entry_point('stcgal==1.4', 'console_scripts', 'stcgal')()
  File "/usr/local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 570, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2751, in load_entry_point
    return ep.load()
  File "/usr/local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2405, in load
    return self.resolve()
  File "/usr/local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2411, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "/usr/local/lib/python3.6/site-packages/stcgal-1.4-py3.6.egg/stcgal/frontend.py", line 27, in <module>
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "/usr/local/lib/python3.6/site-packages/stcgal-1.4-py3.6.egg/stcgal/protocols.py", line 56, in <module>
  File "/usr/local/lib/python3.6/site-packages/stcgal-1.4-py3.6.egg/stcgal/protocols.py", line 71, in StcBaseProtocol
AttributeError: module 'serial' has no attribute 'PARITY_NONE'
`

Any idea for a fix or what could be the reason? I got pyserial installed as well.

zerog2k, 2017-12-11:
are you sure you have pyserial3 installed?
 AttributeError: module 'serial' has no attribute 'PARITY_NONE

not sure but could try:
```
sudo /usr/local/bin/easy_install-3.5 pip
pip3 install pyserial
```

Kyle0122, 2018-11-07:
I am having the same issue.
At first it stuck at "Waiting for MCU, please cycle power:",
and then I installed package serial using pip3,
then now it shows "no attribute 'PARITY_NONE'" every time, even with -h
I tried uninstall package serial but it still says "no attribute 'PARITY_NONE'"

I am using Arch Linux.

grigorig, 2018-11-07:
There's probably a name conflict between the "serial" and "pyserial" modules. I'm not sure how to deal with this. Possibly related: https://github.com/espressif/esptool/issues/269

Kyle0122, 2018-11-08:
Thanks, I fixed the problem by restoring a backup.
but I don't know what to do when it says "Waiting for MCU, please cycle power:"

grigorig, 2018-11-08:
Please check the FAQ.

grigorig, 2018-11-09:
I think it's now clear that this is a module name conflict and it is noted in the FAQ. @Kyle0122 please open a new issue if you continue to have problems.

closed 2018-11-09 by grigorig