grigorig/stcgal #36: Unable to flash STC15F2K08S2

pwsoft, 2017-12-18:
Hi,
Unable to flash STC15F2K08S2 stops at 13% (I assume when it starts writing EEPROM area).

Waiting for MCU, please cycle power: done
Target model:
  Name: STC15F2K08S2
  Magic: F401
  Code flash: 8.0 KB
  EEPROM flash: 53.0 KB
Target frequency: 27.670 MHz
Target BSL version: 7.2.5S
Target wakeup frequency: 35.248 KHz
Target options:
  reset_pin_enabled=False
  clock_source=external
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=3
  eeprom_lvd_inhibit=True
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
  por_reset_delay=short
  rstout_por_state=low
  uart2_passthrough=False
  uart2_pin_mode=normal
  cpu_core_voltage=unknown
Loading flash: 8109 bytes (Intel HEX)
Loading EEPROM: 53993 bytes (Intel HEX)
Option eeprom_erase_enabled=True
Option reset_pin_enabled=True
Option clock_gain=high
Option clock_source=external
Option reset_pin_enabled=False
Switching to 19200 baud: done
Erasing flash: done
Writing flash:  13%|█████████                                                         | 8128/62464 [00:07<00:47, 1146.23 Bytes/s]P
rotocol error: incorrect magic in write packet
Disconnected!
Writing flash:  13%|████████?                                                         | 8192/62464 [00:07<00:47, 1139.93 Bytes/s]

---------------------------------------------------------------------------
Last part of debug dump
Writing flash:  12%|#2        | 7616/62464 [00:06<00:47, 1150.10 Bytes/s]-> Packet data: 46 B9 6A 00 4B 02 1D C0 5A A5 00 08 02 08 F0 01 02 00 00 04 1B 02 09 02 0C 00 03 79 02 07 01 0A C0 C0 CC CC 33 33 CC CC 33 33 02 08 03 01 10 00 00 00 00 CC CC 33 33 FF FE FC F8 F0 E0 C0 80 03 38 03 08 01 1D 18 18 18 1F 1F 13 E3 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
-> Packet data: 46 B9 6A 00 4B 02 1E 00 5A A5 18 18 18 00 00 00 00 0F 0F 0F 0F 18 18 18 1F 1F 00 00 00 00 00 00 F8 F8 03 5D 03 07 03 01 02 07 01 08 1F 1F 18 18 18 18 18 18 03 29 02 08 01 0D FF FF 18 18 18 18 18 18 F8 F8 18 18 18 03 28 03 0B 54 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16

Writing flash:  12%|#2        | 7744/62464 [00:06<00:47, 1150.99 Bytes/s]-> Packet data: 46 B9 6A 00 4B 02 1E 40 5A A5 08 02 08 E0 02 08 07 03 8B 03 0A 03 18 03 0B 01 01 FF 03 80 02 08 01 02 FF FF 03 04 03 08 01 04 0F 0F 0F 0F 03 04 01 07 01 02 F8 F8 03 05 03 07 03 D8 03 0C 01 06 C3 99 91 91 9F 99 04 06 00 00 0E 82 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
-> Packet data: 46 B9 6A 00 4B 02 1E 80 5A A5 04 06 01 00 04 06 02 00 04 06 03 FF 01 7C 62 3C 00 00 00 3C 06 3E 66 3E 00 00 60 60 7C 66 66 7C 00 00 00 3C 60 60 60 3C 00 00 06 06 3E 66 66 3E 00 00 00 3C 66 7E 60 3C 00 00 0E 18 3E 18 18 18 0D 8E 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16

Writing flash:  13%|#2        | 7872/62464 [00:06<00:47, 1148.52 Bytes/s]-> Packet data: 46 B9 6A 00 4B 02 1E C0 5A A5 00 00 00 3E 66 66 3E 06 7C 00 60 60 7C 66 66 66 00 00 18 00 38 18 18 3C 00 00 06 00 06 06 06 06 3C 00 60 60 6C 78 6C 66 00 00 38 18 18 18 18 3C 00 00 00 66 7F 7F 6B 63 00 00 00 7C 66 66 66 66 0F 72 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
-> Packet data: 46 B9 6A 00 4B 02 1F 00 5A A5 00 00 00 3C 66 66 66 3C 00 00 03 7F 00 07 03 38 08 07 01 47 06 00 00 7C 66 60 60 60 00 00 00 3E 60 3C 06 7C 00 00 18 7E 18 18 18 0E 00 00 00 66 66 66 66 3E 00 00 00 66 66 66 3C 18 00 00 00 63 0C A9 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16

Writing flash:  13%|#2        | 8000/62464 [00:06<00:47, 1149.89 Bytes/s]-> Packet data: 46 B9 6A 00 4B 02 1F 40 5A A5 6B 7F 3E 36 00 00 00 66 3C 18 3C 66 00 00 00 66 66 66 3E 0C 78 00 00 7E 0C 18 30 03 D6 00 00 03 D6 01 32 03 08 00 D0 03 D8 02 18 03 30 07 08 01 08 33 99 CC 66 33 99 CC 66 03 00 03 48 01 08 CC 10 B4 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
-> Packet data: 46 B9 6A 00 4B 02 1F 80 5A A5 99 33 66 CC 99 33 66 03 50 03 80 01 07 01 03 06 6C 78 70 60 03 03 03 09 03 E0 03 28 04 08 08 00 03 08 05 00 04 08 0A 00 04 08 0B F8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0A 8A 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16

Writing flash:  13%|#3        | 8128/62464 [00:07<00:47, 1147.75 Bytes/s]-> Packet data: 46 B9 6A 00 4B 02 1F C0 5A A5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 95 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
-> Packet data: 46 B9 6A 00 4B 02 20 00 5A A5 94 E3 7B E3 43 42 4D 42 41 53 49 43 30 A8 41 A7 1D AD F7 A8 A4 AB BE AB 80 B0 05 AC A4 A9 9F A8 70 A8 27 A9 1C A8 82 A8 D1 A8 3A A9 2E A8 4A A9 2C B8 67 E1 55 E1 64 E1 B2 B3 23 B8 7F AA 9F AA 23 89 16
<- Packet data: 46 B9 68 00 08 02 46 00 B8 16
Protocol error: incorrect magic in write packet
-> Packet data: 46 B9 6A 00 07 82 00 F3 16

Writing flash:  13%|#3        | 8192/62464 [00:07<00:47, 1141.84 Bytes/s]

-----------------------------------------------------------------
Any idea what I can do to make it work?
Thanx,
Peter

pwsoft, 2017-12-18:
I've tried both (separate files and combined). Dump shown is when using two separate hex files for code and eeprom space. Command line looked like something like this (but with -D to get debug info):

c:\python36\python.exe stcgal.py -P stc15 -p COM7 -b 19200 -o eeprom_erase_enabled=False -o reset_pin_enabled=False -o clock_gain=high -o clock_source=external -o reset_pin_enabled=False code.hex eepr.hex

I've tried a few combinations with and without -P and -b and various options, but I always got the exact same "Protocol error: incorrect magic in write packet". Looking at the dump data it is exactly offset 0x2000 where the eeprom starts in the STC15F2K08S2.

pan2357, 2021-07-28:
I have found the same problem on STC15W404AS.

From this : https://github.com/grigorig/stcgal/issues/59
Looks like there is some sort of a problem with the EEPROM section.
My temporary solution is to use the official `STC-ISP` to flash a blank code once then use `stcgal` to flash the code normally.

mosagepa, 2021-09-08:
Please try again with proper MAX232-based serial interface/s if any available.
**Also checkout carefully the VCC level, 5V vs 3V3, I had similar problems and they were really caused by that, see e.g.:**
[Trivial solution to "bricked" STC15s](https://github.com/grigorig/stcgal/issues/72#issuecomment-915004682)

Kind regards and good luck!