grigorig/stcgal #40: can not recognition STC89c52

IKTSmart, 2018-07-03:
buring STC89C52RC is ok!
but bring STC89C52 can not recognition, no find STC89C52 device。

grigorig, 2018-07-03:
Hi,

can you provide more detail? Please run stcgal with the `-D` parameter and post the commandline and full log.

IKTSmart, 2018-07-12:
Hi, I sorry,
I try to got more detail.
using:

stcgal.py -p /dev/tty.wchusbserial1450 -a -D
stcgal.py -P auto -p /dev/tty.wchusbserial1450 -a -D
stcgal.py -p /dev/tty.wchusbserial1450 -D

the stcgal unable to continue, stays there：

Cycling power: done
Waiting for MCU: 

Is there any other way to get more information？
I buring with two devices computers, STC89C52RC is OK, STC89C52 stays :Waiting for MCU:

Moufunliux, 2018-10-08:
I met a similar issue too, both on STC89C52 and STC90C516RD+，while downloading the program it just breaks down at Waiting for MCU:
BTW: My system is Ubuntu 18.04.1, and I use  stcgal 1.6.Maybe you should give a detaild supported-device list.

grigorig, 2018-10-08:
There is a list of supported/tested MCU models:

https://github.com/grigorig/stcgal/blob/master/doc/MODELS.md

Unfortunately, there are various possible hardware and software issues that can break things. See the FAQ for mote detail.

ConnorFi, 2019-01-20:
I have noticed that it not support 'STC89C52'
then I quickly buy STC89C52RC😝

yuangu, 2020-02-25:
It's work to change extract_payload function in protocols.py.
form " calc_csum = sum(packet[2:-3]) & 0xff"
to
'''calc_csum = sum(packet[2:-3]) & 0xff"

refer to：
https://github.com/azman/my1stcflash/blob/master/src/my1stc.c

area-8051, 2023-06-01:
Fixed (use stc89a protocol).

IKTSmart, 2023-06-01:
亲，我已收到你的邮件，谢谢！