grigorig/stcgal #43: I/O error: [Errno 5] Input/output error on STC12LE5A60S2

jfrostad, 2018-08-16:
I apologize if opening an issue is not the best way to go about getting support for this package - feel free to redirect me elsewhere if so.

I am having trouble using stcgal on my STC12LE5A60S2, based off instructions given by this software for programming a YQ8003 bike spoke light that has the aforementioned STC12 microcontroller. Please excuse my ignorance as I am new to this technology and seem to be a bit over my head. I am using Ubuntu Linux, and was successfully able to flash new firmware using stcgal several times. However, on the fourth time I got this error: `I/O error: [Errno 5] Input/output error`

I unplugged the USB cable and replugged it back in, then tried again. The firmware flashed successfully, but in all subsequent attempts to flash new firmware I continually get this error and am at my wits end trying to figure out what has happened. Did I somehow brick the microcontroller, as it seems by googling this issue that it is likely a hardware issue.

This is the readout from `dmesg` whenever I try to run any commands on `stcgal`:

```[ 1783.853550] usb 1-4: new full-speed USB device number 12 using xhci_hcd
[ 1784.002836] usb 1-4: New USB device found, idVendor=067b, idProduct=2303
[ 1784.002842] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1784.002846] usb 1-4: Product: USB-Serial Controller
[ 1784.002850] usb 1-4: Manufacturer: Prolific Technology Inc.
[ 1784.003707] pl2303 1-4:1.0: pl2303 converter detected
[ 1784.004660] usb 1-4: pl2303 converter now attached to ttyUSB0
[ 1796.426928] pl2303 1-4:1.0: pl2303_vendor_write - failed to write [0008]: -71
[ 1796.426997] pl2303 1-4:1.0: pl2303_vendor_write - failed to write [0009]: -71
[ 1796.427097] pl2303 ttyUSB0: pl2303_get_line_request - failed: -71
[ 1796.427205] pl2303 ttyUSB0: pl2303_set_line_request - failed: -71
[ 1796.427306] pl2303 1-4:1.0: pl2303_vendor_write - failed to write [0000]: -71
[ 1796.427421] pl2303 ttyUSB0: pl2303_set_control_lines - failed: -71
[ 1796.427584] pl2303 ttyUSB0: pl2303_set_control_lines - failed: -71
[ 1796.707420] pl2303 ttyUSB0: pl2303_set_control_lines - failed: -71
[ 1796.707620] pl2303 ttyUSB0: error sending break = -71
```

Any help would be greatly appreciated as I am clearly out of my depth here.

jfrostad, 2018-08-16:
Here is my readout from stc-isp after checking the MCU:

```Checking target MCU ... 
  MCU type: STC12LE5A60S2
  F/W version: 6.2I

Current H/W Option:
  . Current system clock source is external crystal oscillator
  . Current clock frequency: 21.995MHz
  . Oscillator gain is HIGH
  . RESET pin behaves as reset pin
  . No low level detection on RESET2 pin
  . Power-on reset, use the extra power-on delay
  . Hardware do not enable Watch-Dog-Timer
  . Watch-Dog-Timer pre-scalar : 256
  . Watch-Dog-Timer stop count in idle mode
  . Program can modify the Watch-Dog-Timer scalar
  . Do not detect the level of P1.0 and P1.1 next download
  . Do not erase user EEPROM area at next download

  . MCU type: STC12LE5A60S2
  F/W version: 6.2I

  Complete !(2018-08-16 05:29:53)
```

jfrostad, 2018-08-16:
It doesn't seem that the MCU is bricked as I am able to reprogram it using STC-ISP. Then, I can go back to using stcgal to flash new firmware onto it, but each time around the 2nd or 3rd time I flash it, I start getting this error again. Furthermore, each time the error occurs, I get it one time and then am able to unplug and replug the device, clear the error, flash it once more, and then the error becomes permanent.

jfrostad, 2018-08-18:
Sorry, you can disregard this issue - but i will leave it up in case any other noobs run into it.

The problem ended up being that the only way to power cycle my spoke light is to unplug it and plug it back in. If I run `stcgal` immediately when its plugged in, I am able to access it and flash the firmware.

nomeata, 2020-01-05:
A more reliable way of flashing the chip would be useful. Itt seems that running certain code (tight loops?) in the uploaded program makes it really hard to enter the boot loader.