grigorig/stcgal #46: publishing directly on pypi ?

allan-simon, 2018-09-20:
https://blog.jetbrains.com/pycharm/2017/05/how-to-publish-your-package-on-pypi/

As you can see it's not that much a burden (you create an account on pypi and you launch a simple command to push on their repo)  it will slightly easier to just do `pip install stcgal` especially when including it in bigger project

again thanks a lot for your project :)

grigorig, 2018-09-20:
Yes, this is a good idea. I've been thinking about publishing stcgal on PyPI already and did some cleanups to setup.py a while ago. I'll try to do it in the next few days.
If I don't actually manage to do it, feel free to remind me again. :)

allan-simon, 2018-09-20:
thanks, will do :)

grigorig, 2018-09-24:
stcgal is on PyPI now! After some minor issues with 1.5 released yesterday, 1.6 released just now should work fine.

closed 2018-09-24 by grigorig

allan-simon, 2018-09-24:
hurray \o/

allan-simon, 2018-09-24:
I would put it on your readme that it is as simple as "pip install stcgal"  , as it's really a deal breaker to quickly use it,  personnaly if i see a dedicated link to "installation" i know i'm going to end up with a 10 meter long explanation on how to recompile my kernel