grigorig/stcgal #47: Switching to X baud: Serial port error: read timeout

iainrolland, 2018-10-05:
Have tried switching -b and -l and even -P but if the cycle power down is successful it never moves past the "Switching to X baud: Serial port error: read timeout" error:

stcgal -p /dev/cu.usbserial -P stc12 -b 9600 ledcube8.hex
Waiting for MCU, please cycle power: done
Target model:
  Name: STC12C5A60S2
  Magic: D17E
  Code flash: 60.0 KB
  EEPROM flash: 2.0 KB
Target frequency: 23.098 MHz
Target BSL version: 7.1I
Target options:
  reset_pin_enabled=True
  low_voltage_reset=False
  oscillator_stable_delay=32768
  por_reset_delay=long
  clock_gain=high
  clock_source=external
  watchdog_por_enabled=False
  watchdog_stop_idle=False
  watchdog_prescale=256
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
Loading flash: 3806 bytes (Intel HEX)
Switching to 9600 baud: Serial port error: read timeout

grigorig, 2018-10-06:
Can you post the output with debug mode (`-D`). That will show where exactly it breaks.

iainrolland, 2018-10-08:
Here is output with debug mode on thanks:

Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 3B 50 15 BD 15 BD 15 BD 15 BD 15 BD 15 BD 15 BD 15 BD 71 49 00 D1 7E 8C FF 7F FF FF FF FF FF FF 00 00 00 00 00 00 00 00 00 00 00 00 00 70 80 00 00 C6 74 B0 00 04 14 6E 16
done
Protocol detected: stc12
Target model:
  Name: STC12C5A60S2
  Magic: D17E
  Code flash: 60.0 KB
  EEPROM flash: 2.0 KB
Target frequency: 22.896 MHz
Target BSL version: 7.1I
Target options:
  reset_pin_enabled=True
  low_voltage_reset=False
  oscillator_stable_delay=32768
  por_reset_delay=long
  clock_gain=high
  clock_source=external
  watchdog_por_enabled=False
  watchdog_stop_idle=False
  watchdog_prescale=256
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
Loading flash: 3806 bytes (Intel HEX)
Switching to 9600 baud: -> Packet data: 46 B9 6A 00 0D 50 00 00 36 01 D1 7E 02 4D 16
Serial port error: read timeout

souryou, 2019-06-26:
@grigorig The same situation，I tried -b XX(115200,1200,2400,9600...) but failer

sudo stcgal -P stc15 main.hex -D
Waiting for MCU, please cycle power: <- Packet data: FC
<- Packet data: FF
<- Packet data: FE
<- Packet data: 46 B9 68 00 29 50 FD 18 91 00 00 00 6A 00 01 00 06 00 01 81 D4 04 72 43 00 F1 A5 05 80 00 FF FF FF FF F8 F8 18 04 28 60 0D B2 16
done
Target model:
  Name: STC89C516  ###with 11.0592
  Magic: F1A5
  Code flash: 62.0 KB
  EEPROM flash: 0.0 KB
Target frequency: 0.924 MHz
Target BSL version: 7.2.5C
Target wakeup frequency: 64.792 KHz
Target options:
  reset_pin_enabled=True
  clock_source=external
  clock_gain=high
  watchdog_por_enabled=True
  watchdog_stop_idle=True
  watchdog_prescale=2
  low_voltage_reset=True
  low_voltage_threshold=0
  eeprom_lvd_inhibit=False
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=True
  por_reset_delay=short
  rstout_por_state=high
  uart2_passthrough=True
  uart2_pin_mode=push-pull
Loading flash: 156 bytes (Intel HEX)
Switching to 19200 baud: -> Packet data: 46 B9 6A 00 0E 01 00 40 FF F2 00 00 87 03 31 16
Serial port error: read timeout

grigorig, 2021-01-09:
There have been various timing fixes. Please try again. I'll close this as the issue is very old and the actual problem wasn't found.

closed 2021-01-09 by grigorig