grigorig/stcgal #48: Stuck on Writing Flash:

hwiguna, 2018-11-29:
I'm sure this is user error, but I do not know how to correct it.
I've tried two different hex files and one gets stuck at 75% the other at 92%.
I've tried specifying 9600 baud. No difference.  stuck at same spots.  Please advise. Thank you!

C:\Users\hwiguna\AppData\Local\Programs\Python\Python37-32\Scripts>stcgal -p COM4 -b 9600 "C:\Users\hwiguna\Documents\Projects\Arduino\CubeKitCode\888-Light-Cube-Files_Translated\888-Files\program\Free open source program\7_stc8x8x8.hex"
Waiting for MCU, please cycle power: done
Protocol detected: stc12
Target model:
  Name: STC12C5A60S2
  Magic: D17E
  Code flash: 60.0 KB
  EEPROM flash: 2.0 KB
Target frequency: 23.645 MHz
Target BSL version: 7.1I
Target options:
  reset_pin_enabled=True
  low_voltage_reset=False
  oscillator_stable_delay=32768
  por_reset_delay=long
  clock_gain=high
  clock_source=external
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
Loading flash: 1427 bytes (Intel HEX)
Switching to 9600 baud: testing setting done
Erasing 6 blocks: done
Writing flash:  92%|█████████████████████████████████████████████████████▏    | 1408/1536 [00:19<00:00, 663.74 Bytes/s]

mosagepa, 2021-09-08:
Please try again with proper MAX232-based serial interface/s if any available.
**Also checkout carefully the VCC level, 5V vs 3V3, I had similar problems and they were really caused by that, see e.g.:**
[Trivial solution to "bricked" STC15s](https://github.com/grigorig/stcgal/issues/72#issuecomment-915004682)

Kind regards and good luck!

area-8051, 2023-06-13:
It would be worth testing different protocols (stc12 / stc12a / stc12b) and reopen issue if needed.

closed 2023-06-13 by area-8051