grigorig/stcgal #50: Doesn't work with STC89C52, firmware version 7.2.5C

rgwan, 2019-05-05:
stcgal's stc89 mode only works with STC89C52 4.3C version.
STC89C52 's recent version can't be recognized by stcgal, it shows up an error message:
-----------------------------------------------
Waiting for MCU, please cycle power: done
Protocol detected: stc89
Protocol error: packet checksum mismatch
Disconnected!
-----------------------------------------------
The new STC89 protocol seems to uses 8E1 line coding to transmit data.

Yuns-Wang, 2019-05-29:
I met the some problem too...
Have you solved it yet?

joephon, 2019-06-25:
@rgwan  i am in the same trouble~have you solved it?

histriver, 2019-07-20:
As luck i have two STC, one is STC89C52  7.2C, another is STC90C516RD+ 4.3C. the experiment show that STC90C516RD+ is all right,  the STC89C52 show the same error.  the os is macOS High Sierra 10.13.6.

ChunjieShan, 2020-01-22:
Yeah I've met the same problem and I don't know how to solve that.

yuangu, 2020-02-25:
It's work to change  extract_payload function in protocols.py.
form " calc_csum = sum(packet[2:-3]) & 0xff"
to
'''calc_csum = sum(packet[2:-3]) & 0xff"

grigorig, 2020-02-25:
Both of these snippets are the same, but I think I know what you mean. :)
It should be easy enough to add a special case for newer bootloader versions. Feel free to submit a PR. If I have some free time, I might be able to do it as well, but I cannot test it.

kenyapcomau, 2020-04-29:
I can confirm that using -P stc12 for the 89C52 I got on my development board described in https://github.com/kenyapcomau/QX-mini51 is a valid workaround. Hope this can be automatically detected in future.

nrife, 2021-01-09:
@grigorig 仅更改extract_payload不起作用，7.2.5C与旧版本的通信协议完全不同

nrife, 2021-01-09:
@grigorig Only changing extract_payload will not work, 7.2.5C and the old version of the communication protocol are completely different

grigorig, 2021-01-09:
Yes, that is why I reverted it for now. It looks like newer STC89C52 use an stc12-like protocol and I'd like to avoid the code duplication if possible. I ordered some chips and hopefully I'll get these new ones.

nrife, 2021-01-09:
@grigorig I uploaded the original communication data of 7.2.5c in the fork

area-8051, 2023-06-01:
Fixed (merged nrife's patch).

closed 2023-06-01 by area-8051