grigorig/stcgal #54: STC15F104W - Switching to 19200 baud: Serial port error: read timeout

spth, 2020-06-07:
I use Debian GNU/Linux testing (Python 3) with one of those  CH340G STC15F104W Development Boards.

Output:

~~~~
philipp@notebook5:~/stcgal$ ./stcgal.py -p /dev/ttyUSB2 /tmp/test.ihx
Waiting for MCU, please cycle power: done
Protocol detected: stc15
Target model:
  Name: STC15F104W
  Magic: F294
  Code flash: 4.0 KB
  EEPROM flash: 1.0 KB
Target frequency: 11.051 MHz
Target BSL version: 7.2.5Q
Target wakeup frequency: 36.100 KHz
Target options:
  reset_pin_enabled=False
  clock_source=internal
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=3
  eeprom_lvd_inhibit=True
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart2_passthrough=False
  uart2_pin_mode=normal
  cpu_core_voltage=unknown
Loading flash: 362 bytes (Intel HEX)
Trimming frequency: 11.045 MHz
Switching to 19200 baud: Serial port error: read timeout
~~~~

Actually, as soon as the message "Switching to 19200 baud" appears, the previous program that was on the board when I got it is running again (as I can see from the regular blinking of the LED).

Debug output:

~~~~
philipp@notebook5:~/stcgal$ ./stcgal.py -D -p /dev/ttyUSB2 /tmp/test.ihx
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 34 50 8D 04 8F B2 F7 BB 9F 00 A8 9F 30 FD 27 05 00 00 72 51 03 F2 94 05 06 70 B2 02 23 28 2A 2E 29 80 14 10 04 CA FF 91 BF FF FF 16 09 01 01 11 D6 16
done
Protocol detected: stc15
Target model:
  Name: STC15F104W
  Magic: F294
  Code flash: 4.0 KB
  EEPROM flash: 1.0 KB
Target frequency: 11.051 MHz
Target BSL version: 7.2.5Q
Target wakeup frequency: 36.100 KHz
Target options:
  reset_pin_enabled=False
  clock_source=internal
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=3
  eeprom_lvd_inhibit=True
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart2_passthrough=False
  uart2_pin_mode=normal
  cpu_core_voltage=unknown
Loading flash: 362 bytes (Intel HEX)
Trimming frequency: -> Packet data: 46 B9 6A 00 20 00 0C 00 C0 80 C0 FF C0 00 80 80 80 FF 80 00 40 80 40 FF 40 00 00 80 00 C0 00 0A D3 16
<- Packet data: 46 B9 68 00 20 00 0C 0C 11 11 4D 16 70 18 46 22 B4 2C EB 31 32 45 C8 59 AA 4E 01 6D 46 7B E2 08 B2 16
-> Packet data: 46 B9 6A 00 20 00 0C 8D 80 8E 80 8F 80 90 80 91 80 92 80 8B 40 8C 40 8D 40 8E 40 8F 40 90 40 0B C4 16
<- Packet data: 46 B9 68 00 20 00 0C 23 C2 23 E0 23 F4 23 FE 24 0D 24 26 47 8F 47 B7 47 DF 48 0C 48 39 48 52 09 98 16
11.045 MHz
Switching to 19200 baud: -> Packet data: 46 B9 6A 00 0E 01 8E 40 FB 7F 8F 80 81 04 51 16
<- Packet data: 46 B9 68 00 07 01 00 70 16
-> Packet data: 46 B9 6A 00 0B 05 00 00 5A A5 01 79 16
Serial port error: read timeout
~~~~

spth, 2020-06-07:
For information: On the same system writing on another board, that also uses a CH340G, but with an STC8A8K64S4A12, works.

PSLLSP, 2020-08-16:
Check #17. I think that you have to patch your stcgal...

grigorig, 2021-01-09:
This should work with latest master.

closed 2021-01-09 by grigorig