grigorig/stcgal #56: STC89C52RC Waiting for MCU

Robert-L-Turner, 2020-07-05:
Hello, I've read through the other two issues with no luck, but it is my first time trying to flash an STC chip:

Archlinux with 2 FT232 FTDI USB TTL adapters (1x adafruit FTDI 5v, 1x Aliexpress 5/3.3v). 
Arduino Uno with ISP sketch: http://ceptimus.co.uk/?p=175
Tried Resetting ground
Tried inline resistors on TX/RD
Tried swapping TX>TX/RX>RX
Manually jumping the reset pin to VCC
Restarting the MCU both before and after launching STCGAL 
Tried amending protocal.py as referenced, but suspect this applies only to checksum fails, not the lack of recognition of the MCU

Arduino Uno (lsusb shows it as a FTDI Ltd Bridge) gets a single packet data: F0, vs the two FTDI adapters get nil results.

This https://github.com/ahappyforest/stc-isp says that RS232 is a known issue?

I have a STC15w404as which was immediately recognized by STCGAL using the FTDI adapters.....so appears to be a difference in bootloader with the STC89.

Many thanks in advance,

PSLLSP, 2020-07-22:
It works for me, I have Profilic PL2303 serial adapter and I run it under Linux:

```
user@test:~/Downloads$ stcgal -V
stcgal 1.6

user@test:~/Downloads$ stcgal
Waiting for MCU, please cycle power: done
Protocol detected: stc89
Target model:
  Name: STC89C52RC/LE52R
  Magic: F002
  Code flash: 8.0 KB
  EEPROM flash: 6.0 KB
Target frequency: 11.318 MHz
Target BSL version: 6.6C
Target options:
  cpu_6t_enabled=False
  bsl_pindetect_enabled=False
  eeprom_erase_enabled=False
  clock_gain=high
  ale_enabled=True
  xram_enabled=True
  watchdog_por_enabled=False
Disconnected!
```

PSLLSP, 2020-07-22:
This is the same test, with genuine FTDI FT232RL adapter:

```
user@test:~/Downloads$ stcgal -D
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 3A 00 0A 87 0A 87 0A 87 0A 87 0A 87 0A 87 0A 87 0A 87 66 43 FD F0 02 82 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 46 16
done
Protocol detected: stc89
Target model:
  Name: STC89C52RC/LE52R
  Magic: F002
  Code flash: 8.0 KB
  EEPROM flash: 6.0 KB
Target frequency: 11.088 MHz
Target BSL version: 6.6C
Target options:
  cpu_6t_enabled=False
  bsl_pindetect_enabled=False
  eeprom_erase_enabled=False
  clock_gain=high
  ale_enabled=True
  xram_enabled=True
  watchdog_por_enabled=False
-> Packet data: 46 B9 6A 00 06 82 F2 16
Disconnected!
```

Check serial numbers of your RT232RL adapters, maybe you have FAKE chips. Otherwise share more details about your STC89C52RC chips...

dawendiguo, 2020-10-04:
I use STC89C16  can't connect, But i change STC89C52RC. It work。
I think, there are two possible. one, my STC89C16 is broken. two the "stcgal" only support the chip with "RC" suffix.

area-8051, 2023-06-13:
stcgal working with an STC15**W** (wide supply voltage range) part but not an STC89C (single supply voltage) seems to indicate the voltage on your USB port is a bit too low.

Robert-L-Turner, 2023-06-13:
Thanks alot, I hadn't considered a low supply voltage.  Will check!