grigorig/stcgal #57: STCGAL doesn't work with FAKE chip RT232RL

PSLLSP, 2020-07-22:
There are adapters on the China market that looks like they have genuine RT232RL chip but that is just a FAKE copy of FTDI chip from China. That is not perfect copy and it doesn't work with STCGAL under Linux.

I have adapter with genuine FTDI RT232RL chip and that works great. And I order new one last month from China and it doesn't work with stcgal; It just prints message `Waiting for MCU, please cycle power:` but it doesn't detect that MCU power was "recycled".

One to find FAKE RT232RL? My chips has serial number **A50285BI**, that is well know serial number of fake chips; serial number can be check under Linux with `lsusb -v` command.

It is interesting that FAKE chips work under Windows with STC-ISP, no issue there. I see problem only under Linux. It is possible that there is a problem in Linux kernel and the issue cannot be fixed in stcgal. Maybe that issue is with stcgal and when some timeout will be tuned or some programming sequence slowed down that it will fix compatibility issue with FAKE RT232RL chips...

I have one more adapter that cannot be used with stcgal under Linux, it is based on "cp210x". Maybe that timing in stcgal is just not right...

grigorig, 2020-07-22:
With fake chips, of course, all bets are off. It could be anything. As far as I know, there are several FTDI fakes. I don't know of any specific compatibility issues under Linux, but this is quite possible, of course.

CP2102 works fine for me, though. Do you have some issue with your power cycling maybe? The error points to something like a USB disconnect. Do you see any message about in `dmesg`?

grigorig, 2020-07-22:
Note that it could be phantom power issue with the fake FTDI, too. Maybe the fake chip is able to inject more power through I/O pins.

STC unfortunately uses the worst possible way to switch on the bootloader, causing all kinds of issues.

PSLLSP, 2020-07-23:
CP2102 chip disconnects, it is visible with dmesg. I just cannot use them with stcgal in Linux. I just connect CP2102 adapter to PC, no STC51 is connected and I start "stcgal", message about power is printed. stcgal prints the error in just few seconds, it was re-connected. I do not understand this, it looks like a problem in Linux kernel. Maybe I have "fake" CP2102" chip; I will try to find some older adapter.

I spent some time experimenting with FAKE RT232RL. I think the "problem" is in Linux kernel. If you want to touch these fake chips, just order cheap RT232RL from Aliexpress. I ordered two different adapters a month ago and both have the same "fake" chip.

Both adapters work in Windows with STC-ISP, that is interesting observation and I think it indicate that this is not problem of "phantom power"; I see (RX LED blinks) that chip is rebooted and it sends response but stcgal doesn't receive it or it is "demaged" in some way.

STC-ISP for Windows has many advanced options, one of them is to trigger boot-loader with a custom packet. I do not use this option but it is interesting one from my point of view. I understand it that customer program flashed to the chip monitors serial communication and when it receives "trigger" packet, it reboots the chip and that starts boot-loader.

PSLLSP, 2020-07-23:
I have found older Arduino ISP programmer, it is from "Deek-Robot" and it is based on RT232RL. It is fake chip with famous serial number A50285BI and I didn't know that until today because that ISP works in Linux without issues, like this:

```
[1923312.032360] usb 4-1: new full-speed USB device number 49 using ohci-pci
[1923312.238390] usb 4-1: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[1923312.238395] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1923312.238398] usb 4-1: Product: FT232R USB UART
[1923312.238400] usb 4-1: Manufacturer: FTDI
[1923312.238403] usb 4-1: SerialNumber: A50285BI
[1923312.244476] ftdi_sio 4-1:1.0: FTDI USB Serial Device converter detected
[1923312.244551] usb 4-1: Detected FT232RL
[1923312.250624] usb 4-1: FTDI USB Serial Device converter now attached to ttyUSB0
```

```
$ avrdude -c arduino-ft232r -p m328p
avrdude: AVR device initialized and ready to accept instructions
Reading | ################################################## | 100% 0.10s
avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: safemode: Fuses OK (E:FD, H:DE, L:FF)
avrdude done.  Thank you.
```

I cannot use this adapter with stcgal in Linux, it doesn't work. Maybe that difference is that stcgal uses UART to communicate with the MCU but Arduino ISP uses just bit banging on status lines. I am not sure about this.

PSLLSP, 2020-07-23:
I solved mystery with CP2102. There was other program running in the background that tried to control CP2102 chip. I stopped that daemon and stcgal now works with CP2102 without problem. This was tricky issue... ;-)

PSLLSP, 2020-07-23:
ISP programmer from "deek-robot" uses status lines (CTS, DSR, DCD, RI) to program Ardiuino.
https://arduino.stackexchange.com/questions/30564/ftdi-breakout-with-additional-isp-connector
I can use this adapter with fake RT232RL to program Arduino but it fails when I try to program STC51.

nekromant, 2020-07-24:
> : I/O error: [Errno 5] Input/output error

My lucky guess (and experience with fake nRF24L01) would be that the fake chips may need bigger capacitors on the power rails and making sure that the the voltage drop on the cable isn't that big. 

Another option to try would be that changing the baudrate to an unsupported/high value may trigger some kind of bug in the IC. 

Try using -l and -b options, so that your handshake and upload baudrates match. If these chips work fine with arduino, I would assume the baudrate change is the only really different thing.

PSLLSP, 2020-07-24:
I do not think that this issue has to do anything with parasitic power, filtering capacitors or USB cable. The same configuration works in Windows with STC-ISP SW...

I tried to set the same baud rate but I cannot see any improvement, it just doesn't work...

I have found that when I "cycle" power again and again, some data are received but stcgal doesn't like them. Example, this was produced after about 20 "power cycles":

```
$ stcgal -D -l 2400 -b 2400
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 34 50 31 D2 75 C3 BA BB 7D 01 84 C7 C1 F5 26 BA 00 00 C9 A2 0D 50 AF AC 20 20 DC 3C 02 19 1F 22 91 42 04 10 33 C8 82 DD B1 60 FD FF FD 65 16 AC 20 46
<- Packet data: 09
<- Packet data: E3
<- Packet data: 16
<- Packet data: 46 B9 68 00 28 50 31 D2 75 C3 BA BB 7D 01 42 C7 C1 F5 26 A6 01 02 24 A2 0D 50 AF AC 80 20 DC 3C 02 19 1F 22 24 A1 02 88 06 8D
<- Packet data: B0
<- Packet data: 8B
<- Packet data: 1F
<- Packet data: 4C
<- Packet data: F7
<- Packet data: FD
<- Packet data: 65
<- Packet data: 0A
<- Packet data: AC
<- Packet data: 10
<- Packet data: 46 42
<- Packet data: 96
<- Packet data: F1
<- Packet data: 46 B9 68 00 34 50 31 D2 5D 78 ED BB 3F 01 44 C7 C1 F5 26 A1 00 00 E4 51 03 A8 7D 2B C8 10 6E 9E 02 19 1F 22 24 A1 02 44 06 46 58 8B 1F 4C F7 FD 65 0B AC 10 46 42
<- Packet data: 65
<- Packet data: 16
<- Packet data: 46 B9 68 00 34 50 31 D2 75 C3 BA BB 7D 01 84 C7 C1 F5 26 AB 00 00 C9 A2 0D 50 AF AC 90 20 DC 3C 02 19 1F 22 91 42 04 88 33 64 82 DD EC 60 FD FF FD 65 16 AC 10 46
<- Packet data: 42
<- Packet data: 6A
<- Packet data: 16
```

BTW, I do not like that stcgal doesn't print what (USB) port it is using, even in debug mode. And timestamp are missing too, those could be helpful as the list covers more than 30 seconds of experiments but it is not visible in the list...

This is genuine ST232RL from FTDI, the same chip and the cable:

```
$ stcgal -D -l 2400 -b 2400
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 34 50 8C D2 5D 78 F7 BB 9F 00 A8 FC F0 FD 26 B5 00 00 72 54 00 F5 1F 05 06 70 78 02 19 1F 22 24 28 00 34 20 04 DD EC 60 BF FF FF 19 01 05 30 12 DE 16
done
Protocol detected: stc15
Target model:
  Name: STC15W408AS
  Magic: F51F
  Code flash: 8.0 KB
  EEPROM flash: 5.0 KB
Target frequency: 11.075 MHz
Target BSL version: 7.2.5T
Target wakeup frequency: 36.050 KHz
Target options:
  reset_pin_enabled=False
  clock_source=internal
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=3
  eeprom_lvd_inhibit=True
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart2_passthrough=False
  uart2_pin_mode=normal
  cpu_core_voltage=unknown
-> Packet data: 46 B9 6A 00 07 82 00 F3 16
Disconnected!
```

PSLLSP, 2020-08-24:
I see other compatibility problem at Linux. I tried to use an USB hub and one of them doesn't work under Linux. When serial adapter is connected directly to PC, `stcgal` works without issues. When I insert between PC and serial adapter an USB hub then `stcgal` fails to program chip. I tried with the same USB hub to flash the same chip under Windows (STC-ISP) and there was no such problem.

I tried other USB hub and it worked with `stcgal` without problem. This is strange and it looks like some problem with timing. I am not sure...

The USB hub that doesn't work with `stcgal` is small  3-port USB (2.0) hub with memory card reader. Output from `dmesg`:
```
[909864.266632] usb 2-2: new high-speed USB device number 121 using ehci-pci
[909864.423028] usb 2-2: New USB device found, idVendor=058f, idProduct=6254, bcdDevice= 1.00
[909864.423032] usb 2-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[909864.423340] hub 2-2:1.0: USB hub found
[909864.423398] hub 2-2:1.0: 4 ports detected
[909864.714610] usb 2-2.1: new high-speed USB device number 122 using ehci-pci
[909864.827956] usb 2-2.1: New USB device found, idVendor=058f, idProduct=6366, bcdDevice= 1.00
[909864.827959] usb 2-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[909864.827961] usb 2-2.1: Product: Mass Storage Device
[909864.827963] usb 2-2.1: Manufacturer: Generic
[909864.827964] usb 2-2.1: SerialNumber: 058F63666433
[909864.828386] usb-storage 2-2.1:1.0: USB Mass Storage device detected
[909864.835357] scsi host7: usb-storage 2-2.1:1.0
[909865.867175] scsi 7:0:0:0: Direct-Access     Multiple Card  Reader     1.00 PQ: 0 ANSI: 0
[909865.867458] sd 7:0:0:0: Attached scsi generic sg2 type 0
[909865.892290] sd 7:0:0:0: [sdc] Attached SCSI removable disk
```

nekromant, 2020-08-24:
> The USB hub that doesn't work with stcgal is small 3-port USB hub with memory card reader. Output from dmesg:

Now this _definetely_ looks like power issues.

PSLLSP, 2020-08-24:
> > The USB hub that doesn't work with stcgal is small 3-port USB hub with memory card reader. Output from dmesg:
> 
> Now this _definetely_ looks like power issues. Do you have any leads where I can get some of those fake FT232RL cheaply?

No, it is not power issue. Two different USB hubs with the same PC and the serial adapter but different result. It doesn't look like power issue. Where to get FAKE FT232RL? Any adapter with FT232RL at Aliexpress has FAKE chip... Like this one, https://www.aliexpress.com/item/4000539821134.html

nekromant, 2020-08-24:
Just ordered one. Now I really wonder what those guys could've screwed up in the schematics or silicon so it acts THAT weird. But I'm still pretty sure the symptoms look like power. Why? Because usb hubs have different wiring (cheap thin wires vs more expensive and thicker). If the hub has a cardreader - that also eats some power. If your device works with one hub and doesn't work with the other - in my experience that's pretty much the power issue.

PSLLSP, 2020-08-31:
I ordered "genuine" (not cheap) FT232RL chips to fix my serial adapters with cheap "fake" chips. Unfortunately, new FT232RL chips doesn't work with `stcgal`. It is possible that those are fake chips too but I am not sure because new chips have unique serial numbers. It is possible that these are just "improved" fake chips. :-( FTDI support was not able to confirm if chips are genuine or "fake" chips just on serial number. I ordered my FT232RL chips from distributor that is not on the list of authorized FTDI distributors...

PSLLSP, 2020-08-31:
I tried `stcgal` under Windows. It doesn't work with FAKE chip FT232RL under Windows too! When I try STC-ISP, it works, no problem. So far I assumed that there is some problem in Linux but I can replicate the same under Windows.

I used Win7 box, I installed Python 3 there (https://www.python.org/ftp/python/3.8.5/python-3.8.5-amd64.exe) and than I installed `stcgal` from command line (`pip install stcgal`).

PSLLSP, 2020-09-01:
My genuine FT232RL chip that works great is "revision B" and it was manufactured in year 2010. All other FT232RL adapters I have are with chips in revision C, and those do not work. The key quotation is what is the difference between revision B and C of FT232RL and who is troublemaker, is it `stcgal` or python library?

PSLLSP, 2020-09-01:
I have found another FT232RL adapter in a box with my old projects.It is revison C from year 2014 and it works! So far I have only two adapters that work with stcgal:

OK - A9IXDBBF - 1029-B B9711301
OK - A9WRBTDL - 1403-C CN480661

PSLLSP, 2020-09-01:
I modified `stcgal` to print more details about data it sends and receives. I see difference between genuine and fake chips. When fake chip is used, `stcgal` receives more bytes and it seems that sometimes byte is shifted by one bit (0x05 is sometimes received as 0x0A or 2 bits (0x8F is sometimes 0x3D), etc. My explanation is that STC chips uses baud rate detection and maybe that fake chip doesn't generate perfect baudrate, so STC baudrate is calibrated with serious error.

This is "hello" packet received by real  FT232RL:
```
<- Packet data: 46 B9 68 00 34 50 8F A8 83 A3 F5 7B 9F 00 A8 CE 10 FF 27 19 00 00 72 51 03 F2 94 05 06 70 A3 02 19 1E 21 24 21 80 14 10 04 CC FF 84 BF FF FF 17 07 29 01 12 22 16
```

These packets are received with fake FT232RL, all the same STC chip, the same condition. My STC chip is programmed to auto enter firmware upload (cycle), so I don't need to do any action, I just connect serial adapter and start `stcgal` and I see result.
```
<- Packet data: 46 B9 68 00 68 50 3D A8 83 A3 D5 ED FA 0A 10 75 1E 62 9D F4 08 20 40 39 51 03 C9 94 05 19 82 1B 25 58 CC 28 22 A2 88 C0 14 41 23 8C D6 11 F5 F7 FD 5D 74 2C 59 60 42 0C 16
<- Packet data: 46 B9 68 00 68 50 3D A8 83 A3 D5 ED FA 0A 10 75 1E 62 9D E4 08 20 40 39 51 03 C9 94 05 19 82 1B 25 58 CC 28 22 A2 88 C0 14 41 23 8C D6 11 F5 F7 FD 5D 74 2C 59 60 42 22 F1
<- Packet data: 46 B9 68 00 38 50 3D A8 83 A3 D5 ED FA 05 10 9D 1E B1 9D 02 08 10 40 4E D4 40 F9 94 05 06 C1 1B 25 58 CC 28 22 A2 44 C0 14 41 23 C6 D6 11 FA F7 FD 5D 3A 2C 59 60 42 82 16
```

My theory has one weak point. It doesn't explain why STC-ISP works with fake chips...

PSLLSP, 2020-09-02:
I have found simple 8 channel logical analyzer from "Salease" and checked data on serial line. I understand the issue!
Serial communication of STC15F104W is 8E1 (EVEN PARITY) but `stcgal` configures communication to 8N1 (NONE PARITY). There is a BUG in `stcgal`!

It seem that "fake chips" have serious problem when receiving and handling data with framing error, and framing errors are there when serial data is configured for 8N1 (8-bits) but data are in 8E1 format (9-bits). Fake FT232RL chips synchronize to data stream in different way than genuine FT232RL when UART is configured to 8N1 but data on serial line are in format 8E1. Crystal clear...

PSLLSP, 2020-09-02:
DEMO program for STC MCU for SDCC compiler. It sends data to serial port in format 9600 8E1, just a text strings. It uses software UART and I test it on STC15F104W.

```
#include <stc12.h>

//define baudrate const
//BAUD = 256 - FOSC/3/BAUDRATE/M (1T:M=1; 12T:M=12)
//NOTE: (FOSC/3/BAUDRATE) must be greater than 98, (RECOMMEND GREATER THAN 110)
//#define BAUD 0xF400 // 1200bps @ 11.0592MHz
//#define BAUD 0xFA00 // 2400bps @ 11.0592MHz
//#define BAUD 0xFD00 // 4800bps @ 11.0592MHz
#define BAUD 0xFE80 // 9600bps @ 11.0592MHz
//#define BAUD 0xFF40 //19200bps @ 11.0592MHz
//#define BAUD 0xFFA0 //38400bps @ 11.0592MHz
//#define BAUD 0xEC00 // 1200bps @ 18.432MHz
//#define BAUD 0xF600 // 2400bps @ 18.432MHz
//#define BAUD 0xFB00 // 4800bps @ 18.432MHz
//#define BAUD 0xFD80 // 9600bps @ 18.432MHz
//#define BAUD 0xFEC0 //19200bps @ 18.432MHz
//#define BAUD 0xFF60 //38400bps @ 18.432MHz
//#define BAUD 0xE800 // 1200bps @ 22.1184MHz
//#define BAUD 0xF400 // 2400bps @ 22.1184MHz
//#define BAUD 0xFA00 // 4800bps @ 22.1184MHz
//#define BAUD 0xFD00 // 9600bps @ 22.1184MHz
//#define BAUD 0xFE80 //19200bps @ 22.1184MHz
//#define BAUD 0xFF40 //38400bps @ 22.1184MHz
//#define BAUD 0xFF80 //57600bps @ 22.1184MHz

//sfr AUXR = 0x8E;
//sbit RXB = P3^0; //define UART TX/RX port
//sbit TXB = P3^1;

#define RXB P3_0
#define TXB P3_1
#define BTN P3_2     
#define LED P3_3     

//typedef bit BOOL;
typedef __bit BOOL;
typedef unsigned char BYTE;
typedef unsigned int WORD;
BYTE TBUF,RBUF;
BYTE TDAT,RDAT;
BYTE TCNT,RCNT;
BYTE TBIT,RBIT;
BOOL TING,RING;
BOOL TEND,REND;
BOOL TPAR,RPAR;
void UART_INIT();
BYTE t, r;
BYTE buf[16];

//-----------------------------------------
//Timer interrupt routine for UART
//void tm0() interrupt 1 using 1
void tm0() __interrupt 1 __using 1
{
  if (RING) {
    if (--RCNT == 0) {
      RCNT = 3; //reset send baudrate counter
      if (--RBIT == 0) {
        RBUF = RDAT; //save the data to RBUF
        RING = 0; //stop receive
        REND = 1; //set receive completed flag
      }
      else {
        RDAT >>= 1;
        if (RXB) RDAT |= 0x80; //shift RX data to RX buffer
      }
    }
  }
  else if (!RXB) {
    RING = 1; //set start receive flag
    RCNT = 4; //initial receive baudrate counter
    RBIT = 9; //initial receive bit number (8 data bits + 1 stop bit)
  }

  if (--TCNT == 0) {
    TCNT = 3;    //reset send baudrate counter
    if (TING) {  //judge whether sending
      if (TBIT == 0) {
        TXB = 0; //send start bit
        TDAT = TBUF; //load data from TBUF to TDAT
        TBIT = TDAT ^ (TDAT >> 1);
        TBIT = TBIT ^ (TBIT >> 2);
        TBIT = TBIT ^ (TBIT >> 4);
        TPAR = (TBIT & 1);  // EVEN PARITY

        TBIT = 10; //initial send bit number (8 data bits + parity + 1 stop bit)
      }
    else {
      if (--TBIT == 0) {
        TXB = 1;   //stop bit
        TING = 0; //stop send
        TEND = 1; //set send completed flag
      }
      else if (TBIT == 1) {
        TXB = TPAR;  // PARITY
      }
      else {
        TDAT >>= 1; //shift data to CY
        TXB = CY; //write CY to TX port
      }
    }
  }
}
}

//-----------------------------------------
//initial UART module variable
void UART_INIT()
{
  TING = 0;
  RING = 0;
  TEND = 1;
  REND = 0;
  TCNT = 0;
  RCNT = 0;
}

//-----------------------------------------

void isp_firmware()
{
    //IAP_CONTR |= 1 << 6; // SWBS, activate ISP monitor
    //IAP_CONTR |= 1 << 5; // SWRST, trigger reset
    //IAP_CONTR = (1<<5) | (1<<6);
    IAP_CONTR = 0x60;  // IAP_CONTR is at address 0xC7
}

void putc(const char c)
{
  while (TEND==0);
  TBUF = c;
  TEND = 0;
  TING = 1;
}

void puts(const char *s)
{
   while (*s) putc(*s++);
}

void main()
{

  WORD delay;
  TMOD = 0x00; //timer0 in 16-bit auto reload mode
  AUXR = 0x80; //timer0 working at 1T mode
  TL0 = BAUD & 0xff;
  TH0 = BAUD>>8; //initial timer0 and set reload value
  TR0 = 1;     //tiemr0 start running
  ET0 = 1;     //enable timer0 interrupt
  PT0 = 1;     //improve timer0 interrupt priority
  EA = 1;      //open global interrupt switch

  if (BTN==0) isp_firmware();

  UART_INIT();

  while (1) { //user's function
    puts("FT232RL TEST\n");
    puts("ABCDEFGHIJKLMNOPQRSTUVWXYZ\n");
    puts("0123456789\n\n");
    delay = 30000;
    while (--delay);
    LED ^= 1;
    if (BTN==0) isp_firmware();  // trigger firmware upload with a button
  }

/*
  while (1) { //user's function

    if (BTN==0) isp_firmware();

    if (REND) {
      buf[r++ & 0x0f] = RBUF;
      REND = 0;
    }
    if (TEND) {
      if (t != r) {
        TEND = 0;
        TBUF = buf[t++ & 0x0f];
        TING = 1;
      }
    }
  }
  */
}
```

The program is just source of serial data in 9600 8E1. It is interesting to see how different UARTs see data when configured in wrong way. Try 9600 8E1 and than 9600 8N1.

The FT232RL that I believe is genuine has the same output for "9600 8E1 and 9600 8N1", I cannot see that serial terminal was configured in wrong way. When I try to receive data with "fake" FT232RL, I see garbage when serial terminal is configured for 9600 8N1.

FAKE FT232RL, 9600 8E1:
```
$ picocom -b 9600 --parity e --imap lfcrlf /dev/ttyUSB2
FT232RL TEST
ABCDEFGHIJKLMNOPQRSTUVWXYZ
0123456789
```

FAKE FT232RL, 9600 8N1:
```
$ picocom -b 9600 --parity n --imap lfcrlf /dev/ttyUSB2
�ҪKLMN=��Mm�%�5k�-
0œ6DVdt�����FT232RL TES�S��h�EFGH�S[bk���ԩSTUV]���I�̙3`�dt���H�FT23h�cF��J�
-II�̙345ٺ�����FT23ɓc��J��S��h"EFGH%S[�Z���ԩSQ��e�h�EFGH%S[�Z���ԩSQ���
#U��$R��K�kr��ԩS�����-II�̙345ٺ���H�FT232RL TEST
#*f�$IҥK1kr��ԩSTUY��-II�̙3p�dt���H�FT23
```

genuine FT232RL, 9600 8N1:
```
$ picocom -b 9600 --parity n --imap lfcrlf /dev/ttyUSB2
FT232RL TEST
ABCDEFGHIJKLMNOPQRSTUVWXYZ
0123456789
```

PSLLSP, 2020-09-02:
Test in Python to identify FAKE FT232RL. Similar test like in the previous update, this time is generator written in Python and it switches between "9600 8E1" and "9600 8N1":
```
$ cat uart-test.py

#!/usr/bin/env python3
  
import serial
import time
import sys

BAUDRATE = 9600

if len(sys.argv) < 2:
    print('ERROR, one parameter is required, specify serial port')
    sys.exit(1)

PORT = sys.argv[1]
print('Transmiting serial data to port ' + PORT)

while True:
    with serial.Serial(port=PORT, baudrate=BAUDRATE, parity=serial.PARITY_EVEN, bytesize=serial.EIGHTBITS) as s:
        print('9600 8E1')
        for i in range(3):
            s.write('UART TEST - 9600 8E1: 0123456789 - ABCDEFGHIJKLMNOPQRSTUVWXYZ\n'.encode('utf-8'))
            s.flush()
            time.sleep(1)
    with serial.Serial(port=PORT, baudrate=BAUDRATE, parity=serial.PARITY_NONE, bytesize=serial.EIGHTBITS) as s:
        print('9600 8N1')
        for i in range(3):
            s.write('UART TEST - 9600 8N1: 0123456789 - ABCDEFGHIJKLMNOPQRSTUVWXYZ\n'.encode('utf-8'))
            s.flush()
            time.sleep(1)
```
Some UART is connected to /dev/ttyUSB3:
```
$ python3 uart-test.py /dev/ttyUSB3
Transmiting serial data to port /dev/ttyUSB3
9600 8E1
9600 8N1
9600 8E1
9600 8N1
...
```
Serial adaptor with FT232RL is connected to UART at /dev/ttyUSB2, just three wires (GND-GND, TX-RX, RX-TX), this serial adaptor receives data and it is at /dev/ttyUSB2. This is FAKE FT232RL:
```
$ picocom  -b 9600 --parity n --imap lfcrlf /dev/ttyUSB2

UART TEST - 9600 8N1: 0123456789 - ABCDEFGHIJKLMNOPQRSTUVWXYZ
UART TEST - 9600 8N1: 0123456789 - ABCDEFGHIJKLMNOPQRSTUVWXYZ
2��J� -��dVZ3G�0œ6DVdt��Yd�BCD3�I�JK�kr�J�RSQ���5k�Z
-II�UART TEST - 9600 8N1: 0123456789 - ABCDEFGHIJKLMNOPQRSTUVWXYZ
UART TEST - 9600 8N1: 0123456789 - ABCDEFGHIJKLMNOPQRSTUVWXYZ
UART TEST - 9600 8N1: 0123456789 - ABCDEFGHIJKLMNOPQRSTUVWXYZ
2��J� -��dVZ3G�0123ѫdt�����!CD3�I�JK1k��J�RSQ���5k�Z
-II�� -��dVZ3G�0œ6DVdt��Y��!CDEFG!JM�Z��J�RSQ���
...
```
This is genuine FT232RL or almoust any other serial UART (PL2303, CP2102, etc):
```
$ picocom  -b 9600 --parity n --imap lfcrlf /dev/ttyUSB2

UART TEST - 9600 8N1: 0123456789 - ABCDEFGHIJKLMNOPQRSTUVWXYZ
UART TEST - 9600 8N1: 0123456789 - ABCDEFGHIJKLMNOPQRSTUVWXYZ
UART TEST - 9600 8E1: 0123456789 - ABCDEFGHIJKLMNOPQRSTUVWXYZ
UART TEST - 9600 8E1: 0123456789 - ABCDEFGHIJKLMNOPQRSTUVWXYZ
UART TEST - 9600 8E1: 0123456789 - ABCDEFGHIJKLMNOPQRSTUVWXYZ
UART TEST - 9600 8N1: 0123456789 - ABCDEFGHIJKLMNOPQRSTUVWXYZ
UART TEST - 9600 8N1: 0123456789 - ABCDEFGHIJKLMNOPQRSTUVWXYZ
UART TEST - 9600 8E1: 0123456789 - ABCDEFGHIJKLMNOPQRSTUVWXYZ
...
```

Nable80, 2020-09-02:
I was following this thread for a long time in read-only mode but I can't stop myself anymore, sorry.

Is there any reason to buy FT232 chips nowadays? Especially from Ali, where no one have any reason to provide anything genuine (when cheaper fakes are available).

I'm personally using CH340 USB-UART chips and they're so extremely cheap, that no one makes fake clones of them. At the same time they're good enough for most uses. I agree that you've found real bugs and your research is impressive but is there any reason to insist on using FT232?

PSLLSP, 2020-09-02:
> Is there any reason to buy FT232 chips nowadays? Especially from Ali, where no one have any reason to provide anything genuine (when cheaper fakes are available).

FT232RL has some extra features that are not required for most of projects. When I started to play with STC51 chips I took a "random" adapter from my box with serial adapters. And that was adapter with FT232RL, genuine one, I liked it because connector was well described. Few days later I started other project with STC51 and this time I selected other serial adapter with FT232RL, that one was with fake chip and it was not working. I had two very similar adapters one of them was not working with `stcgal` but both were working in Windows with STCISP. That was a mystery that took my attention...

Results so far:
 - bug in `stcgal` found, it should talk to STC51 in serial format 8E1 (that is why STCISP works)
 - the most advanced Python script to detect fake FTDI chips was published ;-) Man can be surprised how many fake chips are in the World...
 - I have found that fake chips are real business, the advanced fake chips have unique serial numbers and even serious distributor was distributed them as real (they have some official paperwork, probably fake papers that chips are genuine from FTDI factory at Taiwan).

> I'm personally using CH340 USB-UART chips and they're so extremely cheap, that no one makes fake clones of them. At the same time they're good enough for most uses. I agree that you've found real bugs and your research is impressive but is there any reason to insist on using FT232?

I agree, use any serial chip you like, I switched to CH340 chips too, I like mini adapters with micro USB connector (CH340E??).

It is interested that FTDI manufactures even cheaper and simpler UARTs in package with less pins, like FT230X but adapters with these are rare and really expensive. FT232RL was a standard established in the past, it is well known part and it was so popular that fake copies are produced in huge numbers. I had no idea that I have several fake chips in my lab two months ago.

It is normal that manufactures copies popular parts, like timer NE555 but they sell them under their brand. In the case of chips from FTDI, these are produced by pirate manufactures "as real FTDI" but are not 100% compatible, that is bad...

grigorig, 2021-01-03:
> Serial communication of STC15F104W is 8E1 (EVEN PARITY) but `stcgal` configures communication to 8N1 (NONE PARITY). **There is a BUG in `stcgal`!**

Hm... stcgal should use the correct parity configuration. The protocol autodetection might break this, though. Have you tried to manually specify the protocol?

PSLLSP, 2021-01-03:
I have not tried to manually force some protocol, I used auto-detection only.

I have an update to "fake" chips. I bought "genuine" FT232RL and FT232BL from official components distributor and all these chips have the "frame error "bug. My conclusion is that there are no fake chips because the bug is in genuine chips from FTDI. This bug is not documented. The last question that remains is what are FT232RL chips I have that doesn't have the framing error bug. These could be older design from FTDI or fake chips, bad clones of FT232RL... I tried to communicate with FTDI support but I haven't received full explanation of the situation or any plans how to address this issue. They just state that I bought chips from their official distributor ad that implies that my chips are not fake chips.

PSLLSP, 2021-01-03:
I just try FT232RL with framing bug with STC15W201S. When I try autdetect mode, it doesn't work, chips is not detected:
```
$ stcgal -p /dev/ttyUSB0
```
```
Waiting for MCU, please cycle power: 
```
When I force protocol stc15, it works, STC MCU is detected:
```
$ stcgal -P stc15 -p /dev/ttyUSB0
```
```
Waiting for MCU, please cycle power: done
Target model:
  Name: STC15W201S
  Magic: F511
  Code flash: 1.0 KB
  EEPROM flash: 4.0 KB
Target frequency: 11.051 MHz
Target BSL version: 7.2.5T
Target wakeup frequency: 37.715 KHz
Target options:
  reset_pin_enabled=False
  clock_source=internal
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=2
  eeprom_lvd_inhibit=True
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart2_passthrough=False
  uart2_pin_mode=normal
  cpu_core_voltage=unknown
Disconnected!
```

PSLLSP, 2021-01-03:
I just want to highlight that STC-ISP for Windows works, it autodetects chip even when I use FT232RL that has frame error bug...

Just an idea, start communication in 9N1 and autodetect parity in the software... In theory, all communication can be in 9N1 and parity can be handled in SW...

grigorig, 2021-01-03:
Yeah, 8E1 sounds like a good idea, I'm just wary of changing anything as it might affect compatibility with other MCUs or other UART chips and the like. 9N1 is probably not a good idea since support for > 8 bit transfers is spotty (and pyserial doesn't support it at all).

PSLLSP, 2021-01-04:
I see, `stcgal` is written in Python and Serial  library has serious limits what can be configured. `STC-ISP` for Windows can probably do more tricks...
There are reports on internet forums, that it could be possible to trick Serial library to transmit 9N1 data (switch between 8M1 and 8S1) and even possible to receive 9N1 data (set receiver to 8M1 and detect parity errors) but this seems like a dark magic. I have not tried these tricks.

From my point of view, issue #17 is high priority; to support newer chips. I can test it...

grigorig, 2021-01-06:
I did a little bit of testing yesterday, protocol autodetection fails to work with 8E1 on older STC MCUs with protocols that do not natively use parity, such as STC12C2052. So I don't think we should change to that.

Maybe the reception issues on some FT232 can be fixed by setting the number of stop bits to 1.5 or 2. Can you try that? It's a wild guess, but might work. Not sure if that possibly causes new issues, I think the MCUs leave enough space between individual bytes though.

PSLLSP, 2021-01-06:
More stop bits cannot fix bug in FT232RL. I just tried with two stop bits and it doesn't work. Problem is that parity bit zero is used as start bit and triggers reception of new byte, so one or two stop bits doesn't change anything (well, different byte is received). UARTs from other manufactures first wait for stop bit (line in state 1) and only after stop bit is detected, they start to search for start bit, so they synchronize to the next character correctly.

grigorig, 2021-01-09:
The protocol autodetection seems to work fine with MARK parity on allegedly fake FT232 chips. Needs more testing with those older MCUs.

PSLLSP, 2021-01-09:
It is interesting discovery and I can  confirm that it works. I try it with a python script. Transmitter is configured to 8E1. When receiver is 8N1, it receives corrupted data (frame error bug). When receiver is configured to 8M1, 8S1 or even 8O1, it receives data that are synchronized to start bit.  That is really surprise that even 8O1  (odd parity) works...

grigorig, 2021-01-09:
Unfortunately that still doesn't work with old STC MCUs. MARK parity only works reliably if an 8N1 sender (i.e. the older STC MCU in our case) inserts some extra idle time after sending each byte, but at least STC89C52RC doesn't, so the received data is erroneous.

Whatever, it's an acceptable workaround to specify the protocol if you have one of the problematic FT232 UARTs.

closed 2023-06-13 by area-8051