grigorig/stcgal #58: STCGAL fills unused FLASH with 0x00 (should be 0xff)

PSLLSP, 2020-08-17:
When EEPROM file file parameter is passed to `stcgal`, it fills free space between end of program and start of EEPROM area with 0x00 data. When no EEPROM file is passed to `stcgal`, than FLASH is just erased and erase sets unused flash to value 0xff.

```
$ stcgal -V
stcgal 1.6
```

Flash program, no EEPROM (EEPROM area is not erased!):
```
stcgal -p /dev/ttyUSB0 demo.ihx
```

Flash program and EEPROM:
```
stcgal -p /dev/ttyUSB0 demo.ihx eeprom.ihx
```
BTW, I would like to have  a switch that will force stcgal to clear the flash in the chip. I can prepare some hex files to clear the flash but I have to prepare some "dummy" program, etc. 

I will put here a demo program to dump flash memory to serial port....

grigorig, 2021-01-02:
This should be fixed now. Please report back.

grigorig, 2021-01-02:
FWIW, your report contains various suggestions and ideas... please open separate issues if possible.

grigorig, 2021-01-03:
> NOTE: When STC-ISP software detects problem during flashing operation, it sends extra packet `46 B9 6A 00 07 FF 01 70 16` that is never sent from STCGAL (observed during flashing STC15W204S):

Hmm. stcgal currently sends such a packet, but only for the stc8 protocol. Other protocols use the command 0x82 instead of 0xFF. This is what I observed in STC-ISP years ago, but maybe it changed. I always wonder how STC themselves manage this mess of different ISP protocols...

grigorig, 2021-01-09:
Anyway, the initial issue is fixed.

closed 2021-01-09 by grigorig