grigorig/stcgal #59: EEPROM is not erased, cannot be rewritten

PSLLSP, 2020-08-17:
```
$ stcgal -V
stcgal 1.6
```

When I flash EEPROM () second file to `stcgal` to empty chip, it works, no problem. When I try upload EEPROM second time, I receive error `Protocol error: incorrect magic in write packet`. When I check EERPOM area, it was not erased so I thonk that erro means that EEPROM cannot be written because that area is not empty. I can repeat this problem with STC15F104W and STC15W408AS.

```
$ ../stcgal/stcgal.py -p /dev/ttyUSB2 flash-dump.ihx 80.hex 
Waiting for MCU, please cycle power: done
Protocol detected: stc15
Target model:
  Name: STC15W408AS
  Magic: F51F
  Code flash: 8.0 KB
  EEPROM flash: 5.0 KB
Target frequency: 11.021 MHz
Target BSL version: 7.2.5T
Target wakeup frequency: 36.050 KHz
Target options:
  reset_pin_enabled=False
  clock_source=internal
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=3
  eeprom_lvd_inhibit=True
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart2_passthrough=False
  uart2_pin_mode=normal
  cpu_core_voltage=unknown
Loading flash: 662 bytes (Intel HEX)
Loading EEPROM: 1 bytes (Intel HEX)
Trimming frequency: 11.021 MHz
Switching to 19200 baud: done
Erasing flash: done
Writing flash:  93%|████████████████████████████████████████████████████████████████████████████████████      | 8128/8704 [00:08<00:00, 997.68 Bytes/s]Protocol error: incorrect magic in write packet
Disconnected!
Writing flash:  94%|████████████████████████████████████████████████████████████████████████████████████▋     | 8192/8704 [00:08<00:00, 997.12 Bytes/s]
```

flexifi, 2020-09-13:
hi how to get eeprom  to STC MCU

PSLLSP, 2021-01-06:
Could be patch for this issue accepted?

grigorig, 2021-01-06:
I've already noticed this issue and the proposed patch, I'd just like to test it myself with various STC15 series MCUs.

closed 2021-01-09 by grigorig