grigorig/stcgal #61: 65536 byte EEPROM is supported in STC8

lee8871, 2020-12-31:
For STC8 serials,  is available to set all flash as EEPROM(for IAP).
But I can not set "program_eeprom_split" to 65536. 
I just change the max value like this 
``` python
    def set_flash_split(self, val):
        num_val = Utils.to_int(val)
		#lee8871 change - "EEPROM max value to 65536 for stc8"
		if num_val < 512 or num_val > 65536 or (num_val % 512) != 0:
            raise ValueError("must be between 512 and 65536 bytes and a multiple of 512 bytes")
        self.msr[4] = num_val // 256
```
But it is danger for other MCU.  Might you fix this issuse?

grigorig, 2021-01-03:
Do you know why it is dangerous for other chips? Can you brick an MCU with invalid values? I haven't tried invalid values (yet).
If it isn't actually dangerous I would simply relax the check.

lee8871, 2021-02-26:
Thanks  for your repating!
It is worked! I found that I have  made a mistake. It cost time some long. 
While I need an eeprom space 56K. I set ```program_eeprom_split = 57,344```. It did not work. 
When I turn it to ```-o program_eeprom_split=8192^```, It worked.

lee8871, 2021-02-26:
Thanks for your reply!
It is worked! I have made a mistake. It cost time so long.
While I need an eeprom space 56K. I set program_eeprom_split = 57,344. It did not work.
When I turn it to -o program_eeprom_split=8192^, It worked.

closed 2021-02-26 by lee8871