grigorig/stcgal #63: stcgal does not work with STC8G1K

qiwenmin, 2021-01-07:
I have tried stcgal on STC8G1K but it didn't work.

After trying downloading the firmware into STC8G1K chips, these chips could not work even using the official programming tool - stc-isp - anymore.

grigorig, 2021-01-07:
That's bad! I ordered a few of these chips and I'll look into it when I have them. To be honest I think it should not easily be possible to brick a chip. For the meantime, I'm going to add a warning to the README.

qiwenmin, 2021-01-07:
Thanks!

I guess the bootloader of the chip was corrupted.

> That's bad! I ordered a few of these chips and I'll look into it when I have them. To be honest I think it should not easily be possible to brick a chip. For the meantime, I'm going to add a warning to the README.

grigorig, 2021-01-07:
By the way, do you have more information than "didn't work" about what happened when you tried to use stcgal?

qiwenmin, 2021-01-07:
> By the way, do you have more information than "didn't work" about what happened when you tried to use stcgal?

When programming stc8g1k, stcgal seems wait some where until pressing Ctrl+C to terminate. After bricking three chips, I have not try any more.

grigorig, 2021-01-07:
Somewhere? Do you remember when in the programming cycle it stopped? Sorry to hear you bricked so many chips!

closed 2021-01-09 by grigorig

Kashouryo, 2021-01-13:
You might want to pull P3.2 and P3.3 to low and try again. That's a possible reason why the micro doesn't want to enter its bootloader.

qiwenmin, 2021-01-13:
I have tried but it does not work. 😭

> You might want to pull P3.2 and P3.3 to low and try again. That's a possible reason why the micro doesn't want to enter its bootloader. Stcgal might accidentally enable this protection feature.

grigorig, 2021-01-13:
Bricking STC chips can definitely happen. I bricked some STC15W chips by writing incorrect option bytes while working on the stc15 protocol. I'm pretty sure that is what happened here as well.

qiwenmin, 2021-01-13:
Sorry, I made a mistake. In my circuit the power was not completely off when I started programming the chip. I fixed it and it worked with the official programming app.

grigorig, 2021-01-13:
That's good to know!

Would you mind testing programming again with stcgal and posting the output here? Please try with the option `-D` to dump the serial traffic.

qiwenmin, 2021-01-14:
Which protocol should I use with stcgal?

grigorig, 2021-01-14:
Try stc8, i.e. `-P stc8`.

grigorig, 2021-01-14:
Oh, and I think it would be best to use the latest master. Please revert the commit that blacklists STC8G series (`git revert 9900e3088d6f5d33d30a83b9cdac898d363066b7`) beforehand.

qiwenmin, 2021-01-14:
OK. I will try this commit when I am home.

qiwenmin, 2021-01-14:
```bash
➜  stcgal git:(master) python3 stcgal.py -D -P stc8 -p /dev/cu.usbserial-A50285BI -t 6000 -b 2400 ~/devel/ee/stc-dice/firmware/.pio/build/stc-dice/firmware.hex
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 30 50 00 5B 81 00 49 01 03 FF FF FF FF FF 26 B7 F7 73 73 55 00 F7 94 0B 8A C5 A3 CE 0F 1F 20 FF 00 30 00 20 04 A4 20 07 10 01 FF 11 ED 16
done
Target model:
  Name: STC8G1K08A-8PIN
  Magic: F794
  Code flash: 8.0 KB
  EEPROM flash: 4.0 KB
Target frequency: 5.997 MHz
Target BSL version: 7.3.11U
Target wakeup frequency: 35.525 KHz
Target ref. voltage: 1188 mV
Target mfg. date: 2020-07-10
Target options:
  reset_pin_enabled=False
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=False
  low_voltage_threshold=0
  eeprom_erase_enabled=True
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart1_remap=True
  uart2_passthrough=True
  uart2_pin_mode=push-pull
  epwm_open_drain=True
  program_eeprom_split=29440
Loading flash: 794 bytes (Intel HEX)
Trimming frequency: -> Packet data: 46 B9 6A 00 20 00 0C 00 00 17 00 2E 00 45 00 5C 00 73 00 8A 00 A1 00 B8 00 CF 00 E6 00 FF 00 06 86 16
<- Packet data: 46 B9 68 00 20 00 0C 18 05 19 9D 1B 5C 1D 0C 1E B6 20 69 21 F2 23 9C 25 22 26 C0 28 5E 29 F9 08 0B 16
-> Packet data: 46 B9 6A 00 20 00 0C D2 00 D3 00 D4 00 D2 01 D3 01 D4 01 D2 02 D3 02 D4 02 D2 03 D3 03 D4 03 0A 8C 16
<- Packet data: 46 B9 68 00 20 00 0C 27 0B 27 1D 27 0E 27 17 27 29 27 20 27 17 27 11 27 56 26 F0 27 23 27 0E 04 9C 16
5.981 MHz
Switching to 2400 baud: -> Packet data: 46 B9 6A 00 0E 01 00 00 F6 3C 03 D2 80 03 00 16
<- Packet data: 46 B9 68 00 07 01 00 70 16
-> Packet data: 46 B9 6A 00 0B 05 00 00 5A A5 01 79 16
<- Packet data: 46 B9 68 00 07 05 00 74 16
done
Erasing flash: -> Packet data: 46 B9 6A 00 0B 03 00 00 5A A5 01 77 16
Serial port error: read timeout
```

qiwenmin, 2021-01-14:
without `-b 2400`:

```bash
➜  stcgal git:(master) python3 stcgal.py -D -P stc8 -p /dev/cu.usbserial-A50285BI -t 6000 ~/devel/ee/stc-dice/firmware/.pio/build/stc-dice/firmware.hex
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 30 50 00 5B 81 00 49 01 03 FF FF FF FF FF 26 B7 F7 73 73 55 00 F7 94 0B 8A C5 A3 CE 0F 1F 20 FF 00 30 00 20 04 A4 20 07 10 01 FF 11 ED 16
done
Target model:
  Name: STC8G1K08A-8PIN
  Magic: F794
  Code flash: 8.0 KB
  EEPROM flash: 4.0 KB
Target frequency: 5.997 MHz
Target BSL version: 7.3.11U
Target wakeup frequency: 35.525 KHz
Target ref. voltage: 1188 mV
Target mfg. date: 2020-07-10
Target options:
  reset_pin_enabled=False
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=False
  low_voltage_threshold=0
  eeprom_erase_enabled=True
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart1_remap=True
  uart2_passthrough=True
  uart2_pin_mode=push-pull
  epwm_open_drain=True
  program_eeprom_split=29440
Loading flash: 794 bytes (Intel HEX)
Trimming frequency: -> Packet data: 46 B9 6A 00 20 00 0C 00 00 17 00 2E 00 45 00 5C 00 73 00 8A 00 A1 00 B8 00 CF 00 E6 00 FF 00 06 86 16
<- Packet data: 46 B9 68 00 20 00 0C 17 ED 19 94 1B 5C 1D 06 1E 95 20 6F 21 FB 23 8A 25 10 26 D8 28 67 2A 08 07 DE 16
-> Packet data: 46 B9 6A 00 20 00 0C D1 00 D2 00 D3 00 D1 01 D2 01 D3 01 D1 02 D2 02 D3 02 D1 03 D2 03 D3 03 0A 80 16
<- Packet data: 46 B9 68 00 20 00 0C 26 D2 27 23 27 0B 27 05 27 0B 27 05 26 E7 27 02 27 11 26 F3 27 20 27 1A 05 A1 16
5.963 MHz
Switching to 19200 baud: -> Packet data: 46 B9 6A 00 0E 01 00 00 FE C8 00 D1 80 03 90 16
<- Packet data: 46 B9 68 00 07 01 00 70 16
-> Packet data: 46 B9 6A 00 0B 05 00 00 5A A5 01 79 16
<- Packet data: 46 B9 68 00 07 05 00 74 16
done
Erasing flash: -> Packet data: 46 B9 6A 00 0B 03 00 00 5A A5 01 77 16
Serial port error: read timeout
```

grigorig, 2021-01-14:
Okay, that actually looks not too bad! The trickiest part, frequency calibration, seems to work fine. Switching the baudrate also appears to work fine, in both cases. Not sure what seems to go wrong with flash erase, I'll try to debug with my other STC8 MCUs and STC-ISP in the meantime. Thanks for testing!

Kashouryo, 2021-04-08:
Any update on this topic? @grigorig

yuyi1005, 2021-04-22:
There is another problem on STC8G:
With -t 24000, it fails in Trimming frequency.

Trimming frequency: -> Packet data: 46 B9 6A 00 20 00 0C 00 00 17 00 2E 00 45 00 5C 00 73 00 8A 00 A1 00 B8 00 CF 00 E6 00 FF 00 06 86 16
<- Packet data: 46 B9 68 00 20 00 0C 00 86 00 8F 00 98 00 A4 00 AA 00 B6 00 BF 00 C8 00 D1 00 DA 00 E0 00 EC 09 43 16
Protocol error: frequency trimming unsuccessful
-> Packet data: 46 B9 6A 00 07 FF 01 70 16

But with -t 6000 or -t 12000, it will fail in the next step, Erasing flash.

yuyi1005, 2021-04-22:
@qiwenmin @grigorig @Kashouryo 
Hey guys, I have successfully downloaded to STC8G, by changing this place: 

protocols.py (1682)
from: 
`packet += bytes([iap_wait])`
to:
`packet += bytes([0x97])`

I traced the serial streams of STC-ISP, and pinpointed this difference. 

It works with -t 6000 and -t 12000. However, with -t >12000, it fails at Trimming frequency step, which obviously cannot be solved by this amendment. I suppose there is some difference in the fomula of Trimming frequency of STC8G.

xstevexiao, 2022-01-10:
Where did everyone download STC ISP programming software? I tgried http://www.stcmicro.com/rjxz.html then was directed to https://pan.baidu.com/s/1Jm_ll_kaj51qzWihZMq7eQ (code: 67hf) but I don't have a cell phone number in China to set up an account. Thanks.

yuyi1005, 2022-01-10:
> Where did everyone download STC ISP programming software? I tried http://www.stcmicro.com/rjxz.html then was directed to https://pan.baidu.com/s/1Jm_ll_kaj51qzWihZMq7eQ (code: 67hf) but I don't have a cell phone number in China to set up an account. Thanks.

http://www.stcmcudata.com/STCISP/stc-isp-15xx-v6.88L.zip

protobits, 2022-03-06:
> @qiwenmin @grigorig @Kashouryo Hey guys, I have successfully downloaded to STC8G, by changing this place:
>
> protocols.py (1682) from: `packet += bytes([iap_wait])` to: `packet += bytes([0x97])`
>
> I [...]

I've been having the same issue on stc8h1k08 and with this change it now works! Shouldn't this issue be reopened and the proper fix introduced?

eisenmann224, 2022-05-11:
Hi! I'm also interested in using my STC8G & STC8H on Linux. :)
Is there anything I can do to help?

area-8051, 2022-06-10:
This issue is solved by CreativeLau's pull request.
I made it available here: https://github.com/area-8051/stcgal-patched/

area-8051, 2023-06-01:
Fixed (merged stcgal-patched).

commit 9900e308 (2021-01-09): Blacklist STC8G series

stcgal may brick these untested MCUs.

Fixes #63.