grigorig/stcgal #65: [Feature Request] STC U8W Programmer Support

Kashouryo, 2021-01-19:
It'll be nice to have STC's official fancy U8W programmer support. It's an inexpensive dedicated programmer for all STC micros with a few additional features such as offline programming.
[image]

grigorig, 2021-01-20:
I don't know how this device works and I don't have any interest in adding support for it at the moment. If it isn't too much of a break compared to any other protocol support, of course I'm going accept pull requests, though!

area-8051, 2023-06-13:
Among the models imported from STC.ISP.exe are U8-5V and U8-3V.
I assume that these models are used to update the U8 firmware.
Otherwise, the U8W should be seen as a USB-to-serial adapter and work without modification.
The only thing that could need a modification is the automatic MCU power cycling.
You could play with the new -A option to see if it helps.
Could you make a few tests and report your findings?