grigorig/stcgal #66: Doesn't support stc89c52

stan12138, 2021-02-05:
stcgal is a great project, thank you for your work!
But when I try to use it dowmload program for stc89c52 mcu, it doesn't work.
I want know why stcgal doesn't support stc89c52, Is it possible to support stc89c52?
And my second question is how do you find the protocols for these mcu?

nrife, 2021-02-14:
The new version of stc89c52 uses bsl7.2.5, this branch supports it

https://github.com/nrife/stcgal

jefine, 2021-02-23:
Yes, try https://github.com/nrife/stcgal ,and you should use **-P stc89a**

grigorig, 2021-02-23:
FWIW, I finally got my STC89C52 parts with (supposedly) new bootloader. I'll probably look at the code next weekend.

silenceper, 2022-01-26:
@grigorig  any update?

area-8051, 2023-06-01:
Fixed (merged nrife's patch, use stc8a protocol).

closed 2023-06-01 by area-8051