grigorig/stcgal #67: Can't STC8G use the tool?

XiaoPb, 2021-02-25:
Can't STC8G use the tool?
When I used stcgal, I can't download program for my STC8G board.
```
Cycling power: done
Waiting for MCU: done
Protocol detected: stc8     
Target model:
  Name: STC8G1K17-20/16P    
  Magic: F757
  Code flash: 17.0 KB       
  EEPROM flash: 0.0 KB      
Target frequency: 32.009 MHz
Target BSL version: 7.3.10U 
Target wakeup frequency: 35.125 KHz
Target ref. voltage: 1193 mV
Target mfg. date: 2020-03-16
Target options:
  reset_pin_enabled=True
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=0
  eeprom_erase_enabled=True
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart1_remap=False
  uart2_passthrough=True
  uart2_pin_mode=push-pull
  epwm_open_drain=True
  program_eeprom_split=29440
Loading flash: 354 bytes (Intel HEX)
Trimming frequency: Protocol error: frequency trimming unsuccessful
Disconnected!
```

Kashouryo, 2021-03-04:
Grigorig is working on this issue. Probably duplicated with #63.

yuyi1005, 2021-04-22:
https://github.com/grigorig/stcgal/issues/63#issuecomment-824749718

IOsetting, 2021-12-24:
I am working on another tool [stc8prog](https://github.com/IOsetting/stc8prog) for STC8G/STC8H flash programming.

area-8051, 2023-06-01:
Fixed (merged stcgal-patched).

closed 2023-06-01 by area-8051