grigorig/stcgal #68: Does not work with STC15L104W 7.2.5Q

kasbert, 2021-03-31:
I got read timeout in  'Trimming frequency ' with device STC15L104W

Without options. Shouldn't it detect this device ?
```
$ stcgal -D main.ihx 
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 34 50 99 8A E2 4A BC BF 57 0A A0 AB 8B C0 F5 27 05 02 08 20 4E D4 03 C9 D4 15 64 10 6E E1 02 1D 8D 46 51 65 CA 80 14 41 46 98 BD 09 BF FF FD 65 56 D0
<- Packet data: A1
<- Packet data: C0
<- Packet data: 13
<- Packet data: 2E
<- Packet data: 59
<- Packet data: FE
^Cinterrupted
```
As stc15
```
$ stcgal -D -P stc15 main.ihx  
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 34 50 66 71 9C C2 F7 BB 9F 00 B7 2D C0 FD 27 41 00 00 72 51 03 F2 D4 05 06 70 C2 02 1D 23 24 29 29 80 14 10 04 D6 FF C2 BF FF FF 19 05 17 01 13 2E 16
done
Target model:
  Name: STC15L104W
  Magic: F2D4
  Code flash: 4.0 KB
  EEPROM flash: 1.0 KB
Target frequency: 12.005 MHz
Target BSL version: 7.2.5Q
Target wakeup frequency: 26.225 KHz
Target options:
  reset_pin_enabled=False
  clock_source=internal
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=3
  eeprom_lvd_inhibit=True
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart2_passthrough=False
  uart2_pin_mode=normal
  cpu_core_voltage=unknown
Loading flash: 906 bytes (Intel HEX)
Trimming frequency: -> Packet data: 46 B9 6A 00 20 00 0C 00 C0 80 C0 FF C0 00 80 80 80 FF 80 00 40 80 40 FF 40 00 00 80 00 C0 00 0A D3 16
<- Packet data: 46 B9 68 00 20 00 0C 0C 07 11 11 16 39 17 F1 21 F1 2C 00 30 97 44 56 58 29 4D 8E 6C 06 7A A2 07 A9 16
-> Packet data: 46 B9 6A 00 20 00 0C BE 80 BF 80 C0 80 C1 80 C2 80 C3 80 94 40 95 40 96 40 97 40 98 40 99 40 0D 20 16
<- Packet data: 46 B9 68 00 20 00 0C 27 0A 27 19 26 D8 26 F1 27 14 27 19 47 7B 47 A3 47 E4 48 07 48 20 48 43 07 AE 16
12.005 MHz
Switching to 19200 baud: -> Packet data: 46 B9 6A 00 0E 01 97 40 FB 7F C2 80 81 04 8D 16
<- Packet data: 46 B9 68 00 07 01 00 70 16
-> Packet data: 46 B9 6A 00 0B 05 00 00 5A A5 01 79 16
Serial port error: read timeout
```
As stc15a. Target frequency is quite wild.
```
$ stcgal -D -P stc15a main.ihx 
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 34 50 66 71 9C C2 F7 BB 9F 00 B7 2D C0 FD 27 46 00 00 72 51 03 F2 D4 05 06 70 C2 02 1D 23 24 29 29 80 14 10 04 D6 FF C2 BF FF FF 19 05 17 01 13 33 16
done
Target model:
  Name: STC15L104W
  Magic: F2D4
  Code flash: 4.0 KB
  EEPROM flash: 1.0 KB
Target frequency: 175.349 MHz
Target BSL version: 7.2Q
Target options:
  reset_pin_enabled=False
  watchdog_por_enabled=True
[... 156 lines cut ...]
:01004B003282
:030053000200D1D7
:01005B003272
:030063000200F0A8
:0F00C20075316475320075336475340002006661
:030066000202DDB6
:2000D10032C0E0C0D074012527F527E43528F528E43529F529E4352AF52AD0D0D0E032C06E
:2000F10020C0E0C0F0C082C083C007C006C005C004C003C002C001C000C0D075D00885314B
:200111001085321190000975D00012036975D008AE82AF83A234E433F5107511009000647F
:20013100C00FC00E75D00012036975D008AC82AD83D00ED00FEC2EF531ED3FF5328533108B
...
:2002F100B5220280287402B52106E4B5220280337403B52106E4B52202803A8052C235C255
:2003110032758702009000011202CA80C4E52D452E24FF9232E52D452E70B690000212022C
:20033100CA80AED235E52D452E24FF92329000031202CA809CE52D452E24FF9232E52D45F1
:180351002E60030202E29000001202CA0202E29000001202CA0202E275
:06009800E478FFF6D8FD3C
:200076007900E94400601B7A0090038A780175A000E493F2A308B8000205A0D9F4DAF275A3
:02009600A0FFC9
:1D036900E5828510F0A4C582C0F08511F0A4D0F025F0C5838510F0A42583F583223E
:20009E007800E84400600A790175A000E4F309D8FC7800E84400600C7900900001E4F0A360
:0400BE00D8FCD9FA97
:0D006900758134120386E582600302006693
:04038600758200225A
:00000001FF
```

mosagepa, 2021-09-08:
Make sure you're flashing the STC15Ls with 3V3 and not 5V.
If they're on a module designed for any particular purpose, check carefully if there's any regulator or what level the VCC to the chip is really being fed.

See e.g.:**
[Trivial solution to "bricked" STC15s](https://github.com/grigorig/stcgal/issues/72#issuecomment-915004682)

Kind regards and good luck!

kasbert, 2021-09-08:
Well, the programming worked with STC-ISP, so I expect there is a problem with stcgal.

mosagepa, 2021-09-08:
Oh I see. Anyway maybe the info I posted is of some relevance with people having issues.
Good luck, if you find any solution of course let us know

In my case even after noticing the problem was VCC level, I've found that a lot you can learn about STC protocol just by reading and messing a bit with the Python code. Don't be afraid of doing just that (if you have the time, obviously.)

Regards from Spain