grigorig/stcgal #70: Does not work with STC15W4K48S4 (stcgal 1.6)

HookeLiu, 2021-06-04:
I got `Serial port error:read timeout` in `Erasing flash` with device STC15W4K48S4.  
Environment info:  
```log
OS: Windows_NT MinOR-SZM 10.0 19042 x86_64 MS/Windows
USB-UART: USB 串行设备 (COM3) USB\VID_0483&PID_374B&MI_02\6&FD776D8&0&0002 usbser
version: stcgal 1.6
```
LOG:   
```log 
python -m stcgal -p COM3 -P stc15 -D .pio\build\stc15w4k48s4\firmware.hex   
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 34 50 8B BF 68 8B F7 BC 9F 00 A8 E5 80 FD 26 EC 80 00 73 54 00 F5 27 07 06 70 8B 02 17 1C 1F 22 24 00 EC C0 04 E6 EC 6A BF FF FF 20 07 14 80 15 
00 16
done
Target model:
  Name: STC15W4K48S4
  Magic: F527
  Code flash: 48.0 KB
  EEPROM flash: 11.0 KB
Target frequency: 11.069 MHz
Target BSL version: 7.3.7T
Target wakeup frequency: 35.775 KHz
Target options:
  reset_pin_enabled=False
  clock_source=internal
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=4
  eeprom_lvd_inhibit=True
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart2_passthrough=False
  uart2_pin_mode=normal
  cpu_core_voltage=unknown
Loading flash: 165 bytes (Intel HEX)
Trimming frequency: -> Packet data: 46 B9 6A 00 20 00 0C 00 C0 80 C0 FF C0 00 80 80 80 FF 80 00 40 80 40 FF 40 00 00 80 00 C0 00 0A D3 16
<- Packet data: 46 B9 68 00 20 00 0C 0D 1F 12 DD 18 A0 1A 85 25 ED 31 32 35 DD 4B FE 61 20 53 FB 74 BC 84 43 0A 9C 16
-> Packet data: 46 B9 6A 00 20 00 0C 68 80 69 80 6A 80 6B 80 6C 80 6D 80 66 40 67 40 68 40 69 40 6A 40 6B 40 0A 08 16
<- Packet data: 46 B9 68 00 20 00 0C 23 E0 23 F9 24 0D 24 21 24 49 24 5D 47 D0 47 FD 48 1B 48 43 48 6B 48 98 08 F3 16
11.075 MHz
Switching to 19200 baud: -> Packet data: 46 B9 6A 00 0E 01 67 40 FE DF 6A 80 81 04 68 16
<- Packet data: 46 B9 68 00 07 01 00 70 16
-> Packet data: 46 B9 6A 00 0B 05 00 00 5A A5 01 79 16
<- Packet data: 46 B9 68 00 07 05 00 74 16
done
Erasing flash: -> Packet data: 46 B9 6A 00 0B 03 00 00 5A A5 01 77 16
Serial port error: read timeout
```
-------
With stc-isp-15xx-v6.86D.exe  

```log
正在检测目标单片机 ... 
  单片机型号: STC15W4K48S4
  固件版本号: 7.3.7T

当前芯片的硬件选项为:
  . 下次冷启动后系统时钟源为内部IRC振荡器
  . 当前的振荡器的频率: 11.069MHz
  . 振荡器放大增益使能
  . 掉电唤醒定时器的频率: 35.775KHz
  . P3.2和P3.3与下次下载无关
  . 上电复位时增加额外的复位延时
  . 复位引脚用作普通I/O口
  . 检测到低压时复位
  . 低压检测门槛电压 : 2.63 V
  . 低压时不能进行EEPROM操作
  . CPU-Core最高工作电压 : 2.78 V
  . 上电复位时,硬件不启动内部看门狗
  . 上电自动启动内部看门狗时的预分频数为 : 256
  . 空闲状态时看门狗定时器停止计数
  . 启动看门狗后,软件可以修改分频数,但不能关闭看门狗
  . 下次下载用户程序时,不擦除用户EEPROM区
  . 下次下载用户程序时,没有相关的端口控制485
  . 下次下载时不需要校验下载口令
  . TXD与RXD为相互独立的IO
  . 芯片复位后,TXD脚为弱上拉双向口
  . 芯片复位后,P2.0输出高电平
  . 内部安排测试时间: 2020年7月14日

  单片机型号: STC15W4K48S4
  固件版本号: 7.3.7T

开始调节频率 ...			[1.000"]
调节后的频率: 11.069MHz (0.087%)

正在重新握手 ... 成功			[0.109"]
当前的波特率: 115200
正在擦除目标区域 ... 完成 !		[2.344"]
芯片出厂序列号 : F527C5700450F8
正在下载用户代码 ... 完成 !		[0.140"]
正在设置硬件选项 ... 完成 !		[0.031"]

更新后的硬件选项为:
  . 下次冷启动后系统时钟源为内部IRC振荡器
  . 当前的振荡器的频率: 11.069MHz
  . 振荡器放大增益使能
  . P3.2和P3.3与下次下载无关
  . 上电复位时增加额外的复位延时
  . 复位引脚用作普通I/O口
  . 检测到低压时复位
  . 低压检测门槛电压 : 2.63 V
  . 低压时不能进行EEPROM操作
  . CPU-Core最高工作电压 : 2.78 V
  . 上电复位时,硬件不启动内部看门狗
  . 上电自动启动内部看门狗时的预分频数为 : 256
  . 空闲状态时看门狗定时器停止计数
  . 启动看门狗后,软件可以修改分频数,但不能关闭看门狗
  . 下次下载用户程序时,不擦除用户EEPROM区
  . 下次下载用户程序时,没有相关的端口控制485
  . 下次下载时不需要校验下载口令
  . TXD与RXD为相互独立的IO
  . 芯片复位后,TXD脚为弱上拉双向口
  . 芯片复位后,P2.0输出高电平

  . 芯片出厂序列号 : F527C5700450F8
  单片机型号: STC15W4K48S4
  固件版本号: 7.3.7T

  . 用户设定频率: 11.059MHz
  . 调节后的频率: 11.069MHz
  . 频率调节误差: 0.087%

操作成功 !(2021-06-05 00:39:26)
```
I also tried `-P STC15a -b 9600` / `-b 115200` , but I got the same results.

HookeLiu, 2021-06-04:
After I tried many USB-TTL devices, I got success. (the same PC and STC board environment )  
With USB device:  USB-SERIAL CH340 (COM5) USB\VID_1A86&PID_7523\5&5A64A5B&0&1 CH341SER_A64
```log
python -m stcgal -p COM5 -P stc15 -D .pio\build\stc15w4k48s4\firmware.hex
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 34 50 8B BF 68 8B F7 BC 9F 00 A8 FC F0 FD 26 D8 80 00 73 54 00 F5 27 07 06 70 8B 02 17 1C 1F 22 24 00 EC C0 04 E6 EC 6B BF FF FF 20 07 14 80 15 
74 16
done
Target model:
  Name: STC15W4K48S4
  Magic: F527
  Code flash: 48.0 KB
  EEPROM flash: 11.0 KB
Target frequency: 11.075 MHz
Target BSL version: 7.3.7T
Target wakeup frequency: 35.775 KHz
Target options:
  reset_pin_enabled=False
  clock_source=internal
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=4
  eeprom_lvd_inhibit=True
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart2_passthrough=False
  uart2_pin_mode=normal
  cpu_core_voltage=unknown
Loading flash: 165 bytes (Intel HEX)
Trimming frequency: -> Packet data: 46 B9 6A 00 20 00 0C 00 C0 80 C0 FF C0 00 80 80 80 FF 80 00 40 80 40 FF 40 00 00 80 00 C0 00 0A D3 16
<- Packet data: 46 B9 68 00 20 00 0C 0D 1A 12 D3 18 96 1A 76 25 D4 31 19 35 C4 4B D6 60 EE 53 D8 74 7B 83 F3 0B 19 16
-> Packet data: 46 B9 6A 00 20 00 0C 69 80 6A 80 6B 80 6C 80 6D 80 6E 80 67 40 68 40 69 40 6A 40 6B 40 6C 40 0A 14 16
<- Packet data: 46 B9 68 00 20 00 0C 23 E0 23 F9 24 0D 24 35 24 49 24 62 47 D5 47 F3 48 1B 48 43 48 70 48 B1 09 25 16
11.075 MHz
Switching to 19200 baud: -> Packet data: 46 B9 6A 00 0E 01 68 40 FE DF 6B 80 81 04 6A 16
<- Packet data: 46 B9 68 00 07 01 00 70 16
-> Packet data: 46 B9 6A 00 0B 05 00 00 5A A5 01 79 16
<- Packet data: 46 B9 68 00 07 05 00 74 16
done
Erasing flash: -> Packet data: 46 B9 6A 00 0B 03 00 00 5A A5 01 77 16
<- Packet data: 46 B9 68 00 0E 03 F5 27 C5 70 04 50 F8 04 16 16
done
-> Packet data: 46 B9 6A 00 4B 22 00 00 5A A5 02 00 06 02 00 62 75 81 07 12 00 A1 E5 82 60 03 02 00 03 79 00 E9 44 00 60 1B 7A 00 90 00 A5 78 01 75 A0 00 E4 93 F2 A3 08 B8 00 02 05 A0 D9 F4 DA F2 75 A0 FF E4 78 7F F6 D8 FD 78 00 E8 44 00 1B 89 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
Writing flash:   0%|                                                                                                                                                                | 0/512 [00:00<?, ? Bytes/s]-> Packet data: 46 B9 6A 00 4B 02 00 40 5A A5 60 0A 79 01 75 A0 00 E4 F3 09 D8 FC 78 00 E8 44 00 60 0C 79 00 90 00 01 E4 F0 A3 D8 FC D9 FA 02 00 03 75 B4 80 D2 C7 7C 20 7D A1 7E 07 7F 00 1C BC FF 09 1D BD FF 05 1E BE FF 01 1F EC 4D 4E 4F 1E 6F 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
-> Packet data: 46 B9 6A 00 4B 02 00 80 5A A5 70 ED C2 C7 7C 20 7D A1 7E 07 7F 00 1C BC FF 09 1D BD FF 05 1E BE FF 01 1F EC 4D 4E 4F 70 ED 80 C4 75 82 00 22 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 2E 08 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
Writing flash:  38%|███████████████████████████████████████████████████████▌                                                                                            | 192/512 [00:00<00:00, 1717.77 Bytes/s]-> Packet data: 46 B9 6A 00 4B 02 00 C0 5A A5 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 42 36 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
-> Packet data: 46 B9 6A 00 4B 02 01 00 5A A5 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 41 77 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
-> Packet data: 46 B9 6A 00 4B 02 01 40 5A A5 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 41 B7 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
Writing flash:  75%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████                                     | 384/512 [00:00<00:00, 1350.88 Bytes/s]-> Packet data: 46 B9 6A 00 4B 02 01 80 5A A5 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 41 F7 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
-> Packet data: 46 B9 6A 00 4B 02 01 C0 5A A5 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 42 37 16
<- Packet data: 46 B9 68 00 08 02 54 00 C6 16
Writing flash: 576 Bytes [00:00, 1503.97 Bytes/s]
Finishing write: -> Packet data: 46 B9 6A 00 0B 07 00 00 5A A5 01 7B 16
<- Packet data: 46 B9 68 00 08 07 54 00 CB 16
done
Setting options: -> Packet data: 46 B9 6A 00 4B 04 00 00 5A A5 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 FF A8 FF FC FF F0 FF FD FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF EC FF FF FF 6B BF F7 BC 9F 3E 7C 16
<- Packet data: 46 B9 68 00 08 04 54 00 C8 16
done
Target UID: F527C5700450F8
-> Packet data: 46 B9 6A 00 07 82 00 F3 16
Disconnected!
```
It seems STCGAL can't work well with same USB-TTL devices... As all the USB-TTL devices I have tried got success with stc-isp-15xx...

mosagepa, 2021-09-08:
Please try again with proper MAX232-based serial interface/s if any available.
**Also checkout carefully the VCC level, 5V vs 3V3, I had similar problems and they were really caused by that, see e.g.:**
[https://github.com/grigorig/stcgal/issues/72#issuecomment-915004682]

Kind regards and good luck!

closed 2023-06-01 by area-8051