grigorig/stcgal #73: incorrect frame start with STC15W408AS

orzel, 2022-01-14:
Hello. I can't manage to upload a file to my STC15W408AS.

I'm using
* linux
* pyserial 3.5
* stcgal 1.6
* mcu on breadboard, only connecting vcc/gnd/rx/tx

I tried:
* all kind of options -b, -l , -t, as found in various threads/tickets here
* adding 1k resistors on rx/tx
* restarting the MCU by removing/adding gnd instead of vcc
* using a git clone of stcgal instead
* even few patches found on tickers

I have three usb-ttl devices, the result is the same with all of them:
* cp2102
* ft232rl (no idea if it's a fake or not)
* pl2303hx

In all cases, i manage to get the information, but not to send firmware. The mcu otherwise seems to work ok.

Here's a typical test:
```
% stcgal -p /dev/ttyUSB1 m.hex -P stc15 -D
...
<- Packet data: FF
<- Packet data: 46 B9 68 00 34 50 91 50 81 A0 F7 BB 9F 00 A8 B6 A0 FD 25 F2 00 08 72 54 00 F5 1F 05 06 70 A0 02 1C 22 25 27 2B 00 34 20 04 D4 EC 7C BF FF FF 21 04 08 40 12 C3 16
done
Target model:
  Name: STC15W408AS
  Magic: F51F
  Code flash: 8.0 KB
  EEPROM flash: 5.0 KB
Target frequency: 11.057 MHz
Target BSL version: 7.2.5T
Target wakeup frequency: 37.200 KHz
Target options:
  reset_pin_enabled=False
  clock_source=internal
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=3
  eeprom_lvd_inhibit=True
  eeprom_erase_enabled=False
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart2_passthrough=False
  uart2_pin_mode=normal
  cpu_core_voltage=unknown
Loading flash: 6165 bytes (Intel HEX)
Trimming frequency: -> Packet data: 46 B9 6A 00 20 00 0C 00 C0 80 C0 FF C0 00 80 80 80 FF 80 00 40 80 40 FF 40 00 00 80 00 C0 00 0A D3 16
<- Packet data: 00
Protocol error: incorrect frame start
-> Packet data: 46 B9 6A 00 07 82 00 F3 16
Disconnected!
```

So basically, it seems my setup is at least partially ok, stcgal kinda manages to establish a communication with the MCU. I dont get the trimming thinguy (can't even understand what it does).
I'm ready to test whatever you might think worth trying.. But by mysefl, i'm stuck on what to do next.

minlai, 2022-02-04:
Try this [stc8prog](https://github.com/IOsetting/stc8prog)

ceptimus, 2022-06-18:
Having similar issues with an STC15W204S.  stcgal is able to communicate with the device and display the chip information, but the existing program in the chip then starts running while stcgal is 'Trimming frequency:'  The trimming doesn't then happen and stcgal eventually errors out with: Serial port error: read timeout

Linux Mint 20.3 Una base: Ubuntu 20.04 focal
Python 3.8.10
stcgal 1.6

Using a genuine FTDI interface that works fine for uploading to other STC chips: STC8, STC89, etc.

stcgal -P stc15a hourglass.ihx
Waiting for MCU, please cycle power: done
Target model:
  Name: STC15W204S
  Magic: F514
  Code flash: 4.0 KB
  EEPROM flash: 1.0 KB
Target frequency: 177.107 MHz
Target BSL version: 7.2T
Target options:
  reset_pin_enabled=False
  watchdog_por_enabled=True
  watchdog_stop_idle=False
  watchdog_prescale=128
  low_voltage_reset=True
  low_voltage_threshold=0
  eeprom_lvd_inhibit=False
  eeprom_erase_enabled=True
  bsl_pindetect_enabled=True
Loading flash: 739 bytes (Intel HEX)
Trimming frequency: Serial port error: read timeout

orzel, 2022-09-29:
It kinda works now. -P auto wont, but -P stc15 is ok.

I think the original problem lied with the power supply. The MCU is 5V, and at least one of the usb-serial I tried is 5V, but checking with a multimeter, the voltage was more like ~4V once reaching the MCU.

orzel, 2022-09-29:
> Try this [stc8prog](https://github.com/IOsetting/stc8prog)

Did. And it works wonderfully. It autodetects my mcu, while stcgal fails with -P auto (works ok with -P stc15).
Also, I didn't had to specify any 'trimming' parameter, but I guess (?) that maybe that was required only for the first (few) flash burns ?

For the record:
* MCU type: STC15W408AS
* Protocol: STC15B
* F/W version: 7.2.5T

orzel, 2022-09-29:
I asked github to close the ticket but it didn't ?

area-8051, 2023-06-13:
Tested, confirm -P auto uses stc15 protocol.

closed 2023-06-13 by area-8051