grigorig/stcgal #94: Pico probe

kirkegaard, 2023-09-17:
It might be a long shot, but im trying to use stcgal with a pico probe. I can get an output from my target but whenever i try to flash a file i get `Serial port error: pulse timeout`. 

Im using an m2 mac but ive also tried on a pc with same result.
https://github.com/kgsws/mgb_ips_lcd_osd this is the code im trying to flash to a gameboy screen kit.

```
$ stcgal -D -b 9600 -l 9600 -o program_eeprom_split=16384 -P stc8g -p /dev/cu.usbmodem102 flash.ihx
Waiting for MCU, please cycle power: <- Packet data: 46 B9 68 00 38 50 01 6D 5E 60 CD 03 01 FF FF BF AD FF 26 AE F7 3A 73 55 00 F7 57 0C 89 CB 9F CA 0D 1D 1F FF 00 44 00 44 04 A6 20 08 28 32 FF 6E 21 4A 77 92 AD DB FF 16 9E 16
done
Target model:
  Name: STC8G1K17-20/16PIN
  Magic: F757
  Code flash: 17.0 KB
  EEPROM flash: 0.0 KB
Target frequency: 23.945 MHz
Target BSL version: 7.3.12U
Target wakeup frequency: 35.275 KHz
Target ref. voltage: 1190 mV
Target mfg. date: 2020-08-28
Target options:
  reset_pin_enabled=True
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=2
  eeprom_erase_enabled=True
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart1_remap=False
  uart2_passthrough=True
  uart2_pin_mode=push-pull
  epwm_open_drain=True
  program_eeprom_split=14848
Option program_eeprom_split=16384
Loading flash: 13016 bytes (Intel HEX)
Target frequency: -> Packet data: 46 B9 6A 00 18 00 08 00 00 FF 00 00 10 FF 10 00 20 FF 20 00 30 FF 30 05 46 16
Serial port error: pulse timeout
```

grigorig, 2023-09-17:
It looks like picoprobe does not support parity for the UART. This is required for most STC MCUs.

kirkegaard, 2023-09-17:
Alright, i might have guessed :D Thanks for looking into this!

closed 2023-09-17 by kirkegaard

grigorig, 2023-09-18:
It actually shouldn't be too hard to implement parity for picoprobe, maybe you can open an issue for that if you're interested.

kirkegaard, 2023-09-18:
Ooh that would be excellent. Is it a thing for this repo or picoprobe?

grigorig, 2023-09-18:
That would need to be done on picoprobe's side of course.

kirkegaard, 2023-09-18:
Alright! Thanks :)

kirkegaard, 2023-09-18:
Woarh someone created a branch with a possible fix already :o

https://github.com/raspberrypi/picoprobe/tree/line_coding

kirkegaard, 2023-09-18:
https://github.com/grigorig/stcgal/assets/43752/461969b0-552f-4ba1-8784-47e8f86fbd37

It worked!

grigorig, 2023-09-18:
Cool. Nice to see that everything worked out! :)