grigorig/stcgal #95: can this tool flash stc32g12k128 chip?

weiqifa0, 2023-11-03:
hello,this tool can flash stc32g12k128 chip?
or,how can I modify for it ?

weiqifa0, 2023-11-03:
stcgal -P stc8d -b 1500000 -l 2400 -p /dev/tty.usbserial-110 main.hex
Waiting for MCU, please cycle power: done
Target model:
  Name: STC32G12K128
  Magic: F7E3
  Code flash: 128.0 KB
  EEPROM flash: 0.0 KB
Target frequency: 11.064 MHz
Target BSL version: 7.4.5U
Target wakeup frequency: 33.225 KHz
Target ref. voltage: 1189 mV
Target mfg. date: 2022-10-20
Target options:
  reset_pin_enabled=False
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=1
  eeprom_erase_enabled=True
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart1_remap=False
  uart2_passthrough=True
  uart2_pin_mode=push-pull
  epwm_open_drain=True
  program_eeprom_split=0
Loading flash: 1455 bytes (Intel HEX)
Invalid input file. MCU is an MCS-251, input file MUST specify a linear
base address, i.e. contain a type 04 record. More information at:
https://en.wikipedia.org/wiki/Intel_HEX
Disconnected!
(base) crisqifawei@CRISQIFAWEI-MC0 tiny51_os %

weiqifa0, 2023-11-03:
(base) crisqifawei@CRISQIFAWEI-MC0 tiny51_os % 
(base) crisqifawei@CRISQIFAWEI-MC0 tiny51_os % stcgal -v
usage: stcgal [-h] [-e] [-a] [-A {dtr,rts}] [-r RESETCMD]
              [-P {stc89,stc89a,stc12a,stc12b,stc12,stc15a,stc15,stc8,stc8d,stc8g,usb15,auto}] [-p PORT] [-b BAUD]
              [-l HANDSHAKE] [-o OPTION] [-t TRIM] [-D] [-V]
              [code_image] [eeprom_image]
stcgal: error: unrecognized arguments: -v
(base) crisqifawei@CRISQIFAWEI-MC0 tiny51_os % stcgal -V
stcgal 1.8
(base) crisqifawei@CRISQIFAWEI-MC0 tiny51_os %

blu-blue, 2024-01-23:
hello,have you solved this error? I have got the same error as yours

blu-blue, 2024-01-23:
> stcgal -P stc8d -b 1500000 -l 2400 -p /dev/tty.usbserial-110 main.hex Waiting for MCU, please cycle power: done Target model: Name: STC32G12K128 Magic: F7E3 Code flash: 128.0 KB EEPROM flash: 0.0 KB [...]
[image]