grigorig/stcgal #96: falsh Protocol error: uncalibrated, please provide a trim value

weiqifa0, 2023-11-03:
(base) crisqifawei@CRISQIFAWEI-MC0 tiny51_os % make flash
rm -rf *.bin *.hex *.asm *.ihx *.lk *.lst *.map *.mem *.rel *.rst *.sym *.adb *.omf *.cdb
sdcc --model-large --debug -c ./task_scheduling_core/task_scheduling_core.c
sdcc --model-large --debug -c ./chip_platform/platform_interface.c
sdcc --model-large --debug -c ./drivers/lcd1602.c
sdcc --model-large --debug -c ./drivers/uart.c
./drivers/uart.c:8: warning 283: function declarator with no prototype
sdcc --model-large --debug main.c task_scheduling_core.rel platform_interface.rel lcd1602.rel uart.rel
packihx main.ihx > main.hex
packihx: read 67 lines, wrote 113: OK.
stcgal -P stc15 -b 1500000 -l 2400 -p /dev/tty.usbserial-110 main.hex
Waiting for MCU, please cycle power: done
Target model:
  Name: STC15L2K60S2
  Magic: F488
  Code flash: 60.0 KB
  EEPROM flash: 1.0 KB
Target frequency: 0.000 MHz
Target BSL version: 7.2.5S
Target wakeup frequency: 26.975 KHz
Target options:
  reset_pin_enabled=False
  clock_source=internal
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=64
  low_voltage_reset=False
  low_voltage_threshold=3
  eeprom_lvd_inhibit=False
  eeprom_erase_enabled=True
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart2_passthrough=False
  uart2_pin_mode=normal
  cpu_core_voltage=unknown
Loading flash: 1649 bytes (Intel HEX)
Protocol error: uncalibrated, please provide a trim value
Disconnected!
make: *** [flash] Error 1

weiqifa0, 2023-11-03:
(base) crisqifawei@CRISQIFAWEI-MC0 tiny51_os % make flash
rm -rf *.bin *.hex *.asm *.ihx *.lk *.lst *.map *.mem *.rel *.rst *.sym *.adb *.omf *.cdb
sdcc --debug -c ./task_scheduling_core/task_scheduling_core.c
sdcc --debug -c ./chip_platform/platform_interface.c
sdcc --debug -c ./drivers/lcd1602.c
sdcc --debug -c ./drivers/uart.c
./drivers/uart.c:8: warning 283: function declarator with no prototype
sdcc --debug main.c task_scheduling_core.rel platform_interface.rel lcd1602.rel uart.rel
packihx main.ihx > main.hex
packihx: read 61 lines, wrote 101: OK.
stcgal -P stc15 -b 1500000 -l 2400 -D -p /dev/tty.usbserial-110 main.hex
Waiting for MCU, please cycle power: <- Packet data: 00
<- Packet data: 46 B9 68 00 34 50 69 5F 75 96 F5 7B 9F FF FF FF FF FF 26 79 9B B9 72 53 00 F4 88 05 06 70 96 02 0C 11 14 16 1A 00 F4 F0 04 E2 FF 75 BF FF FF 21 05 28 70 17 2A 16
done
Target model:
  Name: STC15L2K60S2
  Magic: F488
  Code flash: 60.0 KB
  EEPROM flash: 1.0 KB
Target frequency: 0.000 MHz
Target BSL version: 7.2.5S
Target wakeup frequency: 26.975 KHz
Target options:
  reset_pin_enabled=False
  clock_source=internal
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=64
  low_voltage_reset=False
  low_voltage_threshold=3
  eeprom_lvd_inhibit=False
  eeprom_erase_enabled=True
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart2_passthrough=False
  uart2_pin_mode=normal
  cpu_core_voltage=unknown
Loading flash: 1460 bytes (Intel HEX)
Protocol error: uncalibrated, please provide a trim value
-> Packet data: 46 B9 6A 00 07 82 00 F3 16
Disconnected!
make: *** [flash] Error 1

grigorig, 2023-11-11:
New MCUs are not calibrated for a specific frequency yet. Like the message says, you need to provide a trim value with -t. Otherwise stcgal doesn't know what clock speed it shall use.