grigorig/stcgal #104: Flashing STC8H1K16 error in stcgal 1.10

Extrunder, 2024-03-17:
stcgal -p COM3 -P stc8 .pio\build\STC8H1K16-rev52\firmware.hex          
Waiting for MCU, please cycle power: done
Target model:
  Name: STC8H1K16
  Magic: F721
  Code flash: 16.0 KB
  EEPROM flash: 12.0 KB
Target frequency: 23.938 MHz
Target BSL version: 7.3.13U
Target wakeup frequency: 35.250 KHz
Target ref. voltage: 1190 mV
Target mfg. date: 2023-01-07
Target options:
  reset_pin_enabled=False
  clock_gain=high
  watchdog_por_enabled=False
  watchdog_stop_idle=True
  watchdog_prescale=256
  low_voltage_reset=True
  low_voltage_threshold=0
  eeprom_erase_enabled=True
  bsl_pindetect_enabled=False
  por_reset_delay=long
  rstout_por_state=high
  uart1_remap=False
  uart2_passthrough=True
  uart2_pin_mode=push-pull
  epwm_open_drain=True
  program_eeprom_split=16384
Loading flash: 1436 bytes (Intel HEX)
Trimming frequency: Protocol error: frequency trimming unsuccessful
Disconnected!

weiszpal, 2024-03-21:
I am experiencing the same issue with STC8G1K08 MCU

Extrunder, 2024-03-22:
Successfully flashed the same setup via STC-ISP