grigorig/stcgal #105: Stc15f104w timer1

raymonds2019, 2024-03-31:
Im sorry if this is unrelated to stcgal but here it goes.

i try a simple timer loop to switch p3.5 at 20ms.

i program my stc15f104e - dip with the -P stc15a and it works perfect, then i try with my stc15f104w - sop -P stc15 and it never works.

Then if i change to timer 0 it works both.

Could there be something i missed?

Olegansv, 2024-04-09:
stc15f104w contains only timer0 and timer2 (no timer1). Also stc15f104w and stc15f104e have different datasheets.

raymonds2019, 2024-04-23:
Hello. Could you provide me with the datasheet for stc15f104w?Sent from my iPhone

Olegansv, 2024-04-24:
[STC15-en.pdf](https://github.com/grigorig/stcgal/files/15099112/STC15-en.pdf)

This datasheet is for all stc15 (excluding early models like stc15f104e).